kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #1: Request/Question

booysenc, 2016-11-30:
Not an issue but wasnt sure how to comment, firstly great job!

Not to the request, is it possible to read the state of a pin using Alexa? for instance if this is connected to a gate and I'd like to know if the gate is open or closed by reading an input if it is high or low

kakopappa, 2016-12-12:
Thanks

Read the state of a pin using Alexa? 
you can check whether pin is high or low like this

```
contact=digitalRead(pin);
if (contact == HIGH) {
 
}
```

Or you can use another sensor to check the status of the device.

booysenc, 2016-12-12:
ok, but how do I ask Alexa to check the pin? Should I define a name like you did contact (which is the variable)
"Alexa what is the status of input 1" or "Alexa please check if the gate is open"... things like that

kakopappa, 2016-12-12:
this is not possible at the moment.

current format for a UPnP device is : 

Alexa, turn [on/off] [device_name]

for any other custom commands like the one you want, you can build them using Alexa Skills Kit
https://developer.amazon.com/public/solutions/alexa/alexa-skills-kit/getting-started-guide

booysenc, 2017-02-08:
Does anyone know how to build such a custom skill, would be nice if alexa could tell me the state of a pin as an input

riban-bw, 2017-11-19:
See issue #25. The WeMo does support request for status but this library does not currently support it. We need someone with a WeMo to record the response to the request so that it could be coded into this library.