kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #3: General info Request

joeman2116, 2016-12-11:
First, thanks for producing some excellent pieces of software.
I run your single Wemo switch software on several esp devices - and it works Flawless!! 
24 hours for a month and the esp8266 units and Alexa  continue to work great.

Now, I was looking at your latest multiple wemo software and have a few questions:

Multiple wemo switch:

1. I am not sure I understand how I use this. Your new software allows one to add several (14) device names and address but It is not clear to me how I would utilize it.  On your single switch if i have a device name" box" and tell Alexa to turn box on/off and  it does. I assign the output pin and it controls a relay  or led etc... perfect!
In the multiple switch I can add several(14) device names with each having a new name and port.
If I were to tell Alexa the device name - say Turn on "Office" or Kitchen etc, What device and or pins does it control...

2. I don't understand how the software controls each device output pin ? and where I put the info.

3. There are several files beside the .ino file, where do I put these in order to have the ide compile? 
As is I get errors when trying to compile.

I guess I need a kick start to get me going on the project....

Thanks for your help

Joe

kakopappa, 2016-12-12:
Thanks 

1. I think you are thinking in terms of sonoff where there is a single relay to control AC. I made this project to control multiple relays connected to a single WeMos D1. For an example, I have a TV and a Fan connected to two power sockets so instead of using 2 WeMos D1s i just wanted to control both using a single WeMos D1. WeMos D1 has multiple IO pins.  When you ask Alexa to discover devices, WeMos D1 will respond back as if two connected.

2. Think you brought a WeMos D1 and a 4-Channel OMRON Solid-State-Relay (https://www.aliexpress.com/item/Sindax-5v-4-Channel-OMRON-SSR-G3MB-202P-Solid-State-Relay-Module-For-Arduino-z3-1pcs/32640442712.html)  This gives you the power to control 4 mains by using 4 I/O pins in D1. So You can define 4 switches in the program to turn high/low accordingly

```
const int relayS1Pin = D1;
const int relayS2Pin = D2;
const int relayS3Pin = D3;
const int relayS4Pin = D4;

// In void setup() 
 pinMode(relayS1Pin, OUTPUT);
 pinMode(relayS2Pin, OUTPUT);
 pinMode(relayS3Pin, OUTPUT);
 pinMode(relayS4Pin, OUTPUT);

...

s1 = new Switch("office", 80, s1On, s1Off);
s2 = new Switch("kitchen", 81, s2On, s2Off);
s3 = new Switch("fan", 82, s3On, s23ff);
s4 = new Switch("tv", 83, s4On, s4Off);

upnpBroadcastResponder.addDevice(*s1);
upnpBroadcastResponder.addDevice(*s2);
upnpBroadcastResponder.addDevice(*s3);
upnpBroadcastResponder.addDevice(*s4);

....

void s1On() {
  digitalWrite(relayS1Pin, HIGH); // turn on relay with voltage HIGH 
}

void s1Off() {
  digitalWrite(relayS1Pin, LOW); // turn on relay with voltage HIGH 
}

void s2On() {

}

void s2Off() {

}

void s3On() {

}

void s3Off() {

}

void s4On() {

}

void s4Off() {

}
```

3. Download the whole folder and open wemos.ino in Arduino IDE. It will open all the other files. Then compile.

joeman2116, 2016-12-12:
That info helped.
I was able to compile and add a new switch --- working very cool!

I was wondering if there any way to see the status of the devices on a browers?

Thanks again...
Great talent , appreciate you sharing.....

Joe

cobra85ec, 2016-12-17:
how do i change the pins i want to control

joeman2116, 2016-12-17:
if you want to keep the same names then change:

in the example

const int relayS1Pin = D1;   change to whatever pin you want to use D2
OR D3 OR D5 etc these represent the gpio pins

joe

cobra85ec, 2016-12-17:
i dont see where that is....

joeman2116, 2016-12-17:
Ok which code are you looking at ? The single or multiple wemo emulator.

Joe

cobra85ec, 2016-12-17:
multiple....

joeman2116, 2016-12-17:
Ok,

You need  6 files:

1. wemos.ino   - this is the main file you edit

2. The other 5 files are support files needed in order to compile.

You  need to go to the github site and download the *wemos directory*

The unzip the files the wemos directory should wemos.ino and contain 5
other files.

Load  / open the wemos.ino using the arduino ide program and then open it
where you stored the wemos,ino

Try to compile as is, if it works then you need to personalize the code for
your pins and names devices etc

Joe

cobra85ec, 2016-12-17:
i did all that... i just dont know where to change the pins... in the single pin ,.... i was able to change it not problem... this one seems a bit harder...

joeman2116, 2016-12-17:
Ok,

So you can compile ok..

I will send you a copy of the working code (.ino I have modified , it
should help... I had  the same issue..

In about 5 minutes
joe

cobra85ec, 2016-12-17:
yes it compiles... and thank u so mach... ill sen u a beer for ur help

joeman2116, 2016-12-17:
// For 3  devices  /// Joe Basque

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"
const int relayPin = D1;// ----------------new
const int relay2Pin = D2;//--------------------new
const int relay5Pin = D5;//-------------------new
//const int flex6Pin = D6;///-------------------new

// prototypes
boolean connectWifi();

//on/off callbacks

void relayOn();//----------------------------new
void relayOff();//---------------------------new
void relay2On();//---------------------------new
void relay2Off();//--------------------------new
void relay5On();//------------------------------new
void relay5Off();//-----------------------------new

// Change this before you flash
const char* ssid = "xxxxxxx";//-------------------new
const char* password = "xxxxxxx";//-----------------new

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *relay = NULL;//--------------NEW D1
Switch *relay2 = NULL;//----------------NEW D2
Switch *relay5= NULL;//-----------------NEW D5

void setup()
{
  Serial.begin(9600);

  // Initialise wifi connection
  wifiConnected = connectWifi();

  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();

    // define output mode
 pinMode(relayPin, OUTPUT);//---------------------------new
 pinMode(relay2Pin, OUTPUT);//-----------------------new
 pinMode(relay5Pin, OUTPUT);//-----------------------------new

    // Define your switches here. Max 14
    // Format: Alexa invocation name, local port no, on callback, off
callback
    //kitchen = new Switch("kitchen lights", 81, kitchenLightsOn,
kitchenLightsOff);
    relay = new Switch("relay  1", 90, relayOn,
relayOff);//---------------------new
    relay2 = new Switch("relay 2", 91, relay2On, relay2Off);
//-------------new
    relay5 = new Switch("relay 5", 92, relay5On, relay5Off);
//----------------------new

    Serial.println("Adding switches upnp broadcast responder");

    upnpBroadcastResponder.addDevice(*relay);//---------new
    upnpBroadcastResponder.addDevice(*relay2);//-----------new
    upnpBroadcastResponder.addDevice(*relay5);//-----------new

  }
}

void loop()
{
if(wifiConnected){
      upnpBroadcastResponder.serverLoop();

      relay->serverLoop();//-----------------new
      relay2->serverLoop();//-------------------new
      relay5->serverLoop();//-------------------new

}
}

void relayOn() {
  digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
-------------new
}

void relayOff() {
  digitalWrite(relayPin, LOW); // turn on relay with voltage HIGH
--------------new
}
void relay2On() {
  digitalWrite(relay2Pin, HIGH); // turn on relay with voltage HIGH
-----------newd2
}

void relay2Off() {
  digitalWrite(relay2Pin, LOW); // turn on relay with voltage HIGH
 ---------- newd2
}

void relay5On() {
  digitalWrite(relay5Pin, HIGH); // turn on relay with voltage HIGH
-----------newd5
}

void relay5Off() {
  digitalWrite(relay5Pin, LOW); // turn on relay with voltage HIGH
 ---------- newd5
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }

  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }

  return state;
}

cobra85ec, 2016-12-17:
this dont compiles

cobra85ec, 2016-12-17:
i change d1 to the pin i wanted and it say callback not defind in scope

joeman2116, 2016-12-17:
//---------->>>>Just did a compile on my latest current working version
//----------->>>>Dont change - do a test compile
//---------->>If it works then change the pins to the ones you wish

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"
const int relayPin = D1;// ----------------new
const int relay2Pin = D2;//--------------------new
const int relay5Pin = D5;//-------------------new
//const int flex6Pin = D6;///-------------------new

// prototypes
boolean connectWifi();

//on/off callbacks

void relayOn();//----------------------------new
void relayOff();//---------------------------new
void relay2On();//---------------------------new
void relay2Off();//--------------------------new
void relay5On();//------------------------------new
void relay5Off();//-----------------------------new
//void flex6On();//------------------------------new
//void flex6Off();//-----------------------------new

// Change this before you flash
const char* ssid = "xxxx";//-------------------new
const char* password = "xxxxxx";//-----------------new

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

//Switch *office = NULL;
//Switch *kitchen = NULL;
Switch *relay = NULL;//--------------NEW D1
Switch *relay2 = NULL;//----------------NEW D2
Switch *relay5= NULL;//-----------------NEW D5
//Switch *flex6 = NULL;//----------------NEW D6

void setup()
{
  Serial.begin(9600);

  // Initialise wifi connection
  wifiConnected = connectWifi();

  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();

    // define output mode
 pinMode(relayPin, OUTPUT);//---------------------------new
 pinMode(relay2Pin, OUTPUT);//-----------------------new
 pinMode(relay5Pin, OUTPUT);//-----------------------------new
// pinMode(flex6Pin, OUTPUT);//-----------------------------new

    // Define your switches here. Max 14
    // Format: Alexa invocation name, local port no, on callback, off
callback
    //kitchen = new Switch("kitchen lights", 81, kitchenLightsOn,
kitchenLightsOff);
    relay = new Switch("relay  1", 90, relayOn,
relayOff);//---------------------new
    relay2 = new Switch("relay 2", 91, relay2On, relay2Off);
//-------------new
    relay5 = new Switch("relay 5", 92, relay5On, relay5Off);
//----------------------new
   // flex6 = new Switch("flex 6", 89, flex6On, flex6Off);
//---------------------new

    Serial.println("Adding switches upnp broadcast responder");

    upnpBroadcastResponder.addDevice(*relay);//---------new
    upnpBroadcastResponder.addDevice(*relay2);//-----------new
    upnpBroadcastResponder.addDevice(*relay5);//-----------new
  //  upnpBroadcastResponder.addDevice(*flex6);//-----------new

  }
}

void loop()
{
if(wifiConnected){
      upnpBroadcastResponder.serverLoop();

      relay->serverLoop();//-----------------new
      relay2->serverLoop();//-------------------new
      relay5->serverLoop();//-------------------new
     // flex6->serverLoop();//-------------------new

}
}

//void officeLightsOn() {
//    Serial.print("Switch 1 turn on ...");
//}

//void officeLightsOff() {
//    Serial.print("Switch 1 turn off ...");
//}

void relayOn() {
  digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH
-------------new
}

void relayOff() {
  digitalWrite(relayPin, LOW); // turn on relay with voltage HIGH
--------------new
}
void relay2On() {
  digitalWrite(relay2Pin, HIGH); // turn on relay with voltage HIGH
-----------newd2
}

void relay2Off() {
  digitalWrite(relay2Pin, LOW); // turn on relay with voltage HIGH
 ---------- newd2
}

void relay5On() {
  digitalWrite(relay5Pin, HIGH); // turn on relay with voltage HIGH
-----------newd5
}

void relay5Off() {
  digitalWrite(relay5Pin, LOW); // turn on relay with voltage HIGH
 ---------- newd5
}

//void flex6On() {
//  digitalWrite(flex6Pin, HIGH); // turn on relay with voltage HIGH
-------------new
//}
//
//void flex6Off() {
//  digitalWrite(flex6Pin, LOW); // turn on relay with voltage HIGH
--------------new
//}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }

  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }

  return state;
}

joeman2116, 2016-12-17:
What esp8266 are you using?
What pin do you want to change?

Joe

cobra85ec, 2016-12-17:
i got it workking,,,, thank u very much......now how do i get u a beer...

joeman2116, 2016-12-17:
Not required but thanks for the offer ,glad to help.....

Joe

cobra85ec, 2016-12-17:
man... im really happy with ur help.....thanks alot

joeman2116, 2016-12-18:
Just got back - blowing the snow out of the  driveway ....

Maybe you will be able to help someone else..

Great program - sure appreciate him releasing it so we all can enjoy it.

Merry Christmas....

73

tvixen, 2017-01-10:
Hi there
Thanks for a great code .... Compilation was working 1.st time with the onboard diodes.

One question ...... When i compile with 11 devices it goes ok every time.... 
Connecting to WiFi
Connecting ......
Connected to WifiTestNet
IP address: 10.0.0.182
RSSI:-77
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 80
WebServer started on port: 81
WebServer started on port: 82
WebServer started on port: 83
WebServer started on port: 84
WebServer started on port: 87
WebServer started on port: 88
WebServer started on port: 89
WebServer started on port: 90
WebServer started on port: 94
WebServer started on port: 95
Adding switches upnp broadcast responder
Adding switch : Samsung T.V. index : 0
Adding switch : T.V. Mute index : 1
Adding switch : T.V. Channel index : 2
Adding switch : T.V. sound index : 3
Adding switch : T.V. one index : 4
Adding switch : Amplifier index : 5
Adding switch : Amplifier sound index : 6
Adding switch : Amplifier aux index : 7
Adding switch : Amplifier tuner index : 8
Adding switch : Amplifier mute index : 9
Adding switch : VacuumCleaner index : 10

BUT
When i compile with 12 devices the witty esp8266 reboot every 10 second. It connect to wifi... and the reboot after 5 sec.
When i compile with 13 devices the witty esp8266 won't connect to wifi and roboot every 5 second
Same goes with devices up to 14 and over. (I could really use 20)
(I have a 100uf over the 3.3V supply for stability)

Have you seen this issue before, or have any idea of what is going wrong ?
Thanks in advance.

dsaleb, 2017-01-14:
Hi Guys, 

I got everything to compile and work except the initial status of the solid state relay for on relay board lights is ON and when I ask alexa to turn each one ON then the on board lights goes off and nothing happen as far as the connected device (it's on all the time). Maybe this Relay is defective? any help appreciated.

Here is my code:

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

const int relayS1Pin = D1;
const int relayS2Pin = D2;
const int relayS3Pin = D3;
const int relayS4Pin = D5;

// prototypes
boolean connectWifi();

//on/off callbacks 
void relayS1On();
void relayS1Off();
void relayS2On();
void relayS2Off();
void relayS3On();
void relayS3Off();
void relayS4On();
void relayS4Off();

// Change this before you flash
const char* ssid = "MyWiFiName";
const char* password = "********";

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *relayS1 = NULL;
Switch *relayS2 = NULL;
Switch *relayS3 = NULL;
Switch *relayS4 = NULL;

void setup()
{     
  Serial.begin(9600);
   
  // Initialise wifi connection
  wifiConnected = connectWifi();
  
  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();
    
     pinMode(relayS1Pin, OUTPUT);
     pinMode(relayS2Pin, OUTPUT);
     pinMode(relayS3Pin, OUTPUT);
     pinMode(relayS4Pin, OUTPUT);
     
    // Define your switches here. Max 14 
    // Format: Alexa invocation name, local port no, on callback, off callback
    relayS1 = new Switch("office lights", 80, relayS1On, relayS1Off);
    relayS2 = new Switch("bedroom lights", 81, relayS2On, relayS2Off);
    relayS3 = new Switch("hallway lights", 82, relayS3On, relayS3Off);
    relayS4 = new Switch("bathroom lights", 83, relayS4On, relayS4Off);
    

    Serial.println("Adding switches upnp broadcast responder");
    upnpBroadcastResponder.addDevice(*relayS1);
    upnpBroadcastResponder.addDevice(*relayS2);
    upnpBroadcastResponder.addDevice(*relayS3);
    upnpBroadcastResponder.addDevice(*relayS4);
  }
}
 
void loop()
{
   if(wifiConnected){
      upnpBroadcastResponder.serverLoop();
      
      relayS1->serverLoop();
      relayS2->serverLoop();
      relayS3->serverLoop();
      relayS4->serverLoop();
   }
}

void relayS1On() {
    digitalWrite(relayS1Pin, HIGH); // turn on relay with voltage HIGH 
}

void relayS1Off() {
    digitalWrite(relayS1Pin, LOW); // turn off relay with voltage HIGH 
}

void relayS2On() {
    digitalWrite(relayS2Pin, HIGH); // turn on relay with voltage HIGH 
}

void relayS2Off() {
    digitalWrite(relayS2Pin, LOW); // turn off relay with voltage HIGH 
}

void relayS3On() {
    digitalWrite(relayS3Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS3Off() {
    digitalWrite(relayS3Pin, LOW); // turn off relay with voltage HIGH 
}
void relayS4On() {
    digitalWrite(relayS4Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS4Off() {
    digitalWrite(relayS4Pin, LOW); // turn off relay with voltage HIGH
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }
  
  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }
  
  return state;
}

VivekESP8266, 2017-03-08:
Hi I too am not able to complie.
Can you please let me know how you resolved the D1 Not declared

kakopappa, 2017-03-08:
@VivekESP8266 https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/14

VivekESP8266, 2017-03-09:
Thank you. 
 Now compiling OK.
However can you please explain what the numbers 90, 91, 92,& 89 point to. 
Also what are the numbers for the other 10 outputs
    relay = new Switch("relay  1", 90, relayOn, relayOff);
    relay2 = new Switch("relay 2", 91, relay2On, relay2Off);
    relay5 = new Switch("relay 5", 92, relay5On, relay5Off);
   // flex6 = new Switch("flex 6", 89, flex6On, flex6Off);

kakopappa, 2017-03-09:
Numbers are refereed to a web server started internally to respond to Alexa for each switch you define. You do not have to worry about them.

VivekESP8266, 2017-03-09:
Dear Aruna
 Thank you for the clarification. So I can use any numbers for it? Or is
there a specific range. As I want to use 8 or 12 mains outputs with solid
state relays.

Best Regards

Vivek Balagopal

+971 50 6540214

labbie48, 2017-12-30:
whenever I go to compile this I get the error:

invalid conversion from 'void (*)()' to 'CallbackFunction {aka bool (*)()}' [-fpermissive]

kakopappa, 2017-12-30:
I am not seeing this error in my Arduino IDE. I can't remember the IDE
version and there is no way to check it because I am going out of city

Try this

https://github.com/witnessmenow/esp8266-alexa-wemo-emulator/tree/master/src

I see they used std::function callback this might fix your problem

labbie48, 2017-12-30:
Maybe someone can see what I missed but when I go to compile it, this is the error I get:
"invalid conversion from 'void ()()' to 'CallbackFunction {aka bool ()()}' [-fpermissive]"

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"
const int relayPin = D1;// ----------------new
const int relay2Pin = D2;//--------------------new
const int relay5Pin = D5;//-------------------new
//const int flex6Pin = D6;///-------------------new

// prototypes
boolean connectWifi();

//on/off callbacks

void relayOn();//----------------------------new
void relayOff();//---------------------------new
void relay2On();//---------------------------new
void relay2Off();//--------------------------new
void relay5On();//------------------------------new
void relay5Off();//-----------------------------new
//void flex6On();//------------------------------new
//void flex6Off();//-----------------------------new

// Change this before you flash
const char* ssid = "xxx";//-------------------new
const char* password = "xxx";//-----------------new

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

//Switch *office = NULL;
//Switch *kitchen = NULL;
Switch *relay = NULL;//--------------NEW D1
Switch *relay2 = NULL;//----------------NEW D2
Switch *relay5= NULL;//-----------------NEW D5
//Switch *flex6 = NULL;//----------------NEW D6

void setup()
{
  Serial.begin(9600);

  // Initialise wifi connection
  wifiConnected = connectWifi();

  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();

    // define output mode
 pinMode(relayPin, OUTPUT);//---------------------------new
 pinMode(relay2Pin, OUTPUT);//-----------------------new
 pinMode(relay5Pin, OUTPUT);//-----------------------------new
// pinMode(flex6Pin, OUTPUT);//-----------------------------new

    // Define your switches here. Max 14
    // Format: Alexa invocation name, local port no, on callback, off callback
    //kitchen = new Switch("kitchen lights", 81, kitchenLightsOn,kitchenLightsOff);
    relay = new Switch("relay 1", 90, relayOn, relayOff);//---------------------new
    relay2 = new Switch("relay 2", 91, relay2On, relay2Off);//-------------new
    relay5 = new Switch("relay 5", 92, relay5On, relay5Off);//----------------------new
   // flex6 = new Switch("flex 6", 89, flex6On, flex6Off);//---------------------new

    Serial.println("Adding switches upnp broadcast responder");

    upnpBroadcastResponder.addDevice(*relay);//---------new
    upnpBroadcastResponder.addDevice(*relay2);//-----------new
    upnpBroadcastResponder.addDevice(*relay5);//-----------new
//  upnpBroadcastResponder.addDevice(*flex6);//-----------new

  }
}

void loop()
{
if(wifiConnected){
      upnpBroadcastResponder.serverLoop();

      relay-&gt;serverLoop();//-----------------new
      relay2-&gt;serverLoop();//-------------------new
      relay5-&gt;serverLoop();//-------------------new
     // flex6-&gt;serverLoop();//-------------------new

}
}

//void officeLightsOn() {
//    Serial.print("Switch 1 turn on ...");
//}

//void officeLightsOff() {
//    Serial.print("Switch 1 turn off ...");
//}

void relayOn() {
  digitalWrite(relayPin, HIGH); // turn on relay with voltage HIGH-------------new
}

void relayOff() {
  digitalWrite(relayPin, LOW); // turn on relay with voltage HIGH--------------new
}
void relay2On() {
  digitalWrite(relay2Pin, HIGH); // turn on relay with voltage HIGH-----------newd2
}

void relay2Off() {
  digitalWrite(relay2Pin, LOW); // turn on relay with voltage HIGH---------- newd2
}

void relay5On() {
  digitalWrite(relay5Pin, HIGH); // turn on relay with voltage HIGH-----------newd5
}

void relay5Off() {
  digitalWrite(relay5Pin, LOW); // turn on relay with voltage HIGH---------- newd5
}

//void flex6On() {
//  digitalWrite(flex6Pin, HIGH); // turn on relay with voltage HIGH-------------new
//}
//
//void flex6Off() {
//  digitalWrite(flex6Pin, LOW); // turn on relay with voltage HIGH--------------new
//}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i &gt; 10){
      state = false;
      break;
    }
    i++;
  }

  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }

  return state;
}

flappypot, 2018-04-20:
I tried to use your code on a nodemcu by modifying the required lines but the relays remain in high state irrespective of the command given. Here is my code:
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// prototypes
boolean connectWifi();

//on/off callbacks 
bool s1On();
bool s1Off();
bool s2On();
bool s2Off();
bool s3On();
bool s3Off();
bool s4On();
bool s4Off();

// Change this before you flash
const char* ssid = "2ndFLOOR";
const char* password = "books4ume";
const int relayS1Pin = 12;
const int relayS2Pin = 14;
const int relayS3Pin = 13;
const int relayS4Pin = 15;

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *s1 = NULL;
Switch *s2 = NULL;
Switch *s3 = NULL;
Switch *s4 = NULL;

bool iss1On = false;
bool iss2On = false;
bool iss3On = false;
bool iss4On = false;

void setup()
{
  Serial.begin(9600);
   
  // Initialise wifi connection
  wifiConnected = connectWifi();
  
  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();
    
  
  // Inside void setup() 
  pinMode(relayS1Pin, OUTPUT);
  pinMode(relayS2Pin, OUTPUT);
  pinMode(relayS3Pin, OUTPUT);
  pinMode(relayS4Pin, OUTPUT);
    // Define your switches here. Max 10
    // Format: Alexa invocation name, local port no, on callback, off callback
    s1 = new Switch("office", 81, s1On, s1Off);
    s2 = new Switch("kitchen", 90, s2On, s2Off);
    s3 = new Switch("fan", 91, s3On, s3Off);
    s4 = new Switch("tv", 92, s4On, s4Off);

    Serial.println("Adding switches upnp broadcast responder");
  upnpBroadcastResponder.addDevice(*s1);
  upnpBroadcastResponder.addDevice(*s2);
  upnpBroadcastResponder.addDevice(*s3);
  upnpBroadcastResponder.addDevice(*s4);

  }
}
 
void loop()
{
   if(wifiConnected){
      upnpBroadcastResponder.serverLoop();
      
      s1->serverLoop();
      s2->serverLoop();
      s3->serverLoop();
      s4->serverLoop();

      
   }
}

bool s1On() {
    Serial.println("Switch 1 turn on ...");
    
    iss1On = true;    
    return iss1On;
}

bool s1Off() {
    Serial.println("Switch 1 turn off ...");

    iss1On = false;
    return iss1On;
}

bool s2On() {
    Serial.println("Switch 2 turn on ...");

    iss2On = true;
    return iss2On;
}

bool s2Off() {
  Serial.println("Switch 2 turn off ...");

  iss2On = false;
  return iss2On;
}

bool s3On() {
    Serial.println("Switch 3 turn on ...");

    iss3On = true;
    return iss3On;
}

bool s3Off() {
  Serial.println("Switch 3 turn off ...");

  iss3On = false;
  return iss3On;
}

bool s4On() {
    Serial.println("Switch 4 turn on ...");

    iss4On = true;
    return iss4On;
}

bool s4Off() {
  Serial.println("Switch 4 turn off ...");

  iss4On = false;
  return iss4On;
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }
  
  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }
  
  return state;
}

kakopappa, 2018-04-20:
You have to change the relay status to high or low in on/off functions

flappypot, 2018-04-20:
Thanks for your reply, I am not a programmer so could you just point out exactly what lines need to be modified. That would be great!

flappypot, 2018-04-22:
@kakopappa 

Thanks for your help. The code is working beautifully !!