kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #5: Question:  WeMo Smart LED Bulbs

tvixen, 2017-01-04:
Hi
Thanks for a great piece of code...total cool, im using it everyday.

One question, do you plan to implement the WeMo Smart LED Bulbs ?
So it will be possible to dim the bulb, or set anything else to a certain value...volume, or channel up/down ?

Thanks in advance.

kakopappa, 2017-01-05:
Thanks. 

I have not looked at WeMo Smart LED Bulbs yet mainly, because they are expensive (at least for me).  I assume they work some what similar. Try sniffing on HTTP traffic from Alexa to bulb and see. 

I have different setup at home .. a cheap one.  I brought milights from AliExpress about 10$ each. Then brought the remote controller about 7$ and used my raspberry pi + rf controller to emulate the remote.  then used fauxmo (https://github.com/makermusings/fauxmo) to trun on/off using alexa. I have been using this setup for about 3 months now, didn't burn down my house yet.

https://www.raspberrypi.org/magpi/pi-mi-light/

tvixen, 2017-01-05:
Ok, sounds great with your setup. Always funny to make new things.
I read that Alexa do not support Wemo Light Bulb, so I think I will look at something else to copy a up/down function from. 

I need a function to control all this: 

/* Declare Samsung TV callbacks  (on-off)  */
void SamsungTVOn();         //Samsung TV toggle ON/OFF
void SamsungTVOff();        //Samsung TV toggle ON/OFF
void SamsungMuteOn();       //Samsung TV Mute
void SamsungChannelUp();    //Samsung TV Channal up
void SamsungChannelDown();  //Samsung TV Channal Down
void SamsungVolumeUp();     //Samsung TV Volume up
void SamsungVolumeDown();   //Samsung TV Volume down
void SamsungCH1On();        //Samsung TV Ch 1
void SamsungCH2On();        //Samsung TV Ch 2
void SamsungCH3On();        //Samsung TV Ch 3

/* Declare Marantz callbacks  (on-off) */
void AmplifierOn();           //Amplifier ON
void AmplifierOff();          //Amplifier OFF
void AmplifierVolumeUp();     //Amplifier volume up
void AmplifierVolumeDown();   //Amplifier volume down
void AmplifierAuxOn();        //Amplifier Aux input
void AmplifierTunerOn();      //Amplifier Tuner input
void AmplifierTvOn();         //Amplifier TV input
void AmplifierDVDOn();        //Amplifier DVD input
void AmplifierScourceUp();    //Amplifier Source up
void AmplifierScourceDown();  //Amplifier Source down
void AmplifierMuteOn();       //Amplifier Mute toggle
void AmplifierMuteOff();      //Amplifier Mute toggle

/* Declare Mile callbacks  (on-off) */
void VacuumCleanerOn();       //Vaccum Cleaner toggle ON
void VacuumCleanerCornerOn(); //Corners ON
void VacuumCleanerGoOn();     //Go Go Go
void VacuumCleanerBaseOn();   //Return to home

As you can see, I need use a couples of up/down functions to control volume aso.  So if you have any great ideas, I would gladly hear them.

tvixen, 2017-01-06:
This seems to be a winner 
https://www.engadget.com/2017/01/03/wemo-dimmer-switch-and-mini/

tvixen, 2017-01-07:
Btw do you have issue with the discovery numbers. Sometimes Alexa will only discover 4 switches, even if theres 8 ?  And I have to discover 3-4 times.