kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #6: Crash when more than 11 devices

tvixen, 2017-01-11:
Hi there
And thanks for a great code .... Compilation was working 1.st time with the onboard diodes.

One question ...... When i compile with 11 devices it goes ok every time....
(heres the connect log)

Connecting to WiFi
Connecting ......
Connected to WifiTestNet
IP address: 10.0.0.182
RSSI:-77
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 80
WebServer started on port: 81
WebServer started on port: 82
WebServer started on port: 83
WebServer started on port: 84
WebServer started on port: 87
WebServer started on port: 88
WebServer started on port: 89
WebServer started on port: 90
WebServer started on port: 94
WebServer started on port: 95
Adding switches upnp broadcast responder
Adding switch : Samsung T.V. index : 0
Adding switch : T.V. Mute index : 1
Adding switch : T.V. Channel index : 2
Adding switch : T.V. sound index : 3
Adding switch : T.V. one index : 4
Adding switch : Amplifier index : 5
Adding switch : Amplifier sound index : 6
Adding switch : Amplifier aux index : 7
Adding switch : Amplifier tuner index : 8
Adding switch : Amplifier mute index : 9
Adding switch : VacuumCleaner index : 10

BUT
When I compile with 12 devices the witty esp8266 reboot every 10 second. 
It connect to wifi... and the reboot after 5 sec.

When i compile with 13 devices the witty esp8266 won't connect to wifi 
and roboot every 5 second.

Same goes with devices up to 14 and over. (I could really use 20)
(I have a 100uf over the 3.3V supply for stability)

**Have you seen this issue before, or have any idea of what is going wrong ?**
Thanks in advance.

kakopappa, 2017-01-12:
that's a lot of switches... I have not tried using more than 3 or 4. 

have you looked at the crash log? 
https://github.com/me-no-dev/EspExceptionDecoder

tvixen, 2017-01-12:
Ok...seems like a great tool. I will try it out and see if I can track it..... thanks.

laz2wiringpi, 2017-01-16:
i have found a similar issue ... 
random works for 4  or rarely 5 devices 

when i added a define DEBUG to switch.h 
and  
        #if defined(DEBUG)
                   Serial.print*(" XXXX");   
         #endif 
 to remove serial.* code ...

the issue so far  seems to work  ...  and i can add 6 devices 

is suspect its the time for processing serial data and or out of  memory.
 

and it would seem there are a lot of flavors of the esp boards .  
  Adafruit HUZZAH ESP8266 breakout.

tvixen, 2017-01-16:
Yes there's a lots of brands. Mine is the Witty ESP8266-12F  board.
Did you try without your laptop. Just with a powersupply around 2A.

laz2wiringpi, 2017-01-16:
Not power issue I use a lab power supply.The issue is the server loop in main Each Web server can't respond while handling another request  Modified the payload to min bare bones Setup.xml without debug serial works 
Best just to use more than one device past 5 to 6 ports 

Sent via the Samsung Galaxy S7, an AT&T 4G LTE smartphone

  more than 11 devices (#6)

tvixen, 2017-01-16:
Well it shouldn't be necessary to do two things at the same time. as Alexa can only call call one webserver port at the time. 
1. Ask Alexa
2. Alexa send the request to port 80
3. ESP8266, Do the request
4. ESP8266 go back to listen
Repeat------
2. Alexa send the request to port 81
3. ESP8266, Do the request
4. ESP8266 go back to listen

laz2wiringpi, 2017-01-16:
>    5. Alexa send the request to port 81
>    6. ESP8266, Do the request
>    7. ESP8266 go back to listen

It's the discovery ... setup.xml

I think it polls every so often not positive though

tvixen, 2017-01-16:
Remember to delete Alexa devices in your app, before you scan for new devices.  Remember to ask Alexa, and do NOT scan via the app.