kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #7: All Lights On instead of Off

dsaleb, 2017-01-14:
Hi Guys,

I got everything to compile and work except the initial status of the relay board lights is ON and when I ask alexa to turn each one ON then the on board lights goes off and nothing happened as far as the connected device (it's on all the time). I tried 2 different relays, any help appreciated.

Here is my code:

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include 
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

const int relayS1Pin = D1;
const int relayS2Pin = D2;
const int relayS3Pin = D3;
const int relayS4Pin = D5;

// prototypes
boolean connectWifi();

//on/off callbacks
void relayS1On();
void relayS1Off();
void relayS2On();
void relayS2Off();
void relayS3On();
void relayS3Off();
void relayS4On();
void relayS4Off();

// Change this before you flash
const char* ssid = "MyWiFiName";
const char* password = "********";

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *relayS1 = NULL;
Switch *relayS2 = NULL;
Switch *relayS3 = NULL;
Switch *relayS4 = NULL;

void setup()
{
Serial.begin(9600);

// Initialise wifi connection
wifiConnected = connectWifi();

if(wifiConnected){
upnpBroadcastResponder.beginUdpMulticast();

 pinMode(relayS1Pin, OUTPUT);
 pinMode(relayS2Pin, OUTPUT);
 pinMode(relayS3Pin, OUTPUT);
 pinMode(relayS4Pin, OUTPUT);
 
// Define your switches here. Max 14 
// Format: Alexa invocation name, local port no, on callback, off callback
relayS1 = new Switch("office lights", 80, relayS1On, relayS1Off);
relayS2 = new Switch("bedroom lights", 81, relayS2On, relayS2Off);
relayS3 = new Switch("hallway lights", 82, relayS3On, relayS3Off);
relayS4 = new Switch("bathroom lights", 83, relayS4On, relayS4Off);

Serial.println("Adding switches upnp broadcast responder");
upnpBroadcastResponder.addDevice(*relayS1);
upnpBroadcastResponder.addDevice(*relayS2);
upnpBroadcastResponder.addDevice(*relayS3);
upnpBroadcastResponder.addDevice(*relayS4);
}
}

void loop()
{
if(wifiConnected){
upnpBroadcastResponder.serverLoop();

  relayS1->serverLoop();
  relayS2->serverLoop();
  relayS3->serverLoop();
  relayS4->serverLoop();
}
}

void relayS1On() {
digitalWrite(relayS1Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS1Off() {
digitalWrite(relayS1Pin, LOW); // turn off relay with voltage HIGH
}

void relayS2On() {
digitalWrite(relayS2Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS2Off() {
digitalWrite(relayS2Pin, LOW); // turn off relay with voltage HIGH
}

void relayS3On() {
digitalWrite(relayS3Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS3Off() {
digitalWrite(relayS3Pin, LOW); // turn off relay with voltage HIGH
}
void relayS4On() {
digitalWrite(relayS4Pin, HIGH); // turn on relay with voltage HIGH
}

void relayS4Off() {
digitalWrite(relayS4Pin, LOW); // turn off relay with voltage HIGH
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
boolean state = true;
int i = 0;

WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
Serial.println("");
Serial.println("Connecting to WiFi");

// Wait for connection
Serial.print("Connecting ...");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
if (i > 10){
state = false;
break;
}
i++;
}

if (state){
Serial.println("");
Serial.print("Connected to ");
Serial.println(ssid);
Serial.print("IP address: ");
Serial.println(WiFi.localIP());
}
else {
Serial.println("");
Serial.println("Connection failed.");
}

return state;
}

joeman2116, 2017-01-15:
Code looks ok to me.
Wondering what you are using for relays?
Are they high level or low level triggers?

Joe

dsaleb, 2017-01-15:
Hi Joe,

I tried two different relays

https://www.aliexpress.com/item/Sindax-5v-4-Channel-OMRON-SSR-G3MB-202P-Solid-State-Relay-Module-For-Arduino-z3-1pcs/32640442712.html

and

https://www.aliexpress.com/item/5V-8-Channel-Relay-Module-Controller-For-Arduino-Mega2560-UNO-R3-Raspberry-Pi/32705468846.html?

joeman2116, 2017-01-15:
Sorry but cant determine if they are high or low level triggers.
Relays come in high an low level triggers. Some are both depending on a
jumper.
You could just take 5 volts to power the relays , then apply 5 volts to a
channel input to see if relays turns on/off as you apply the 5 volts.
I know i have purchased which were supposed to be high trigger levels only
to find out they were low.
Joe

dsaleb, 2017-01-16:
I tried to switch the HIGH and LOW in the code but still didn't work.

Ex.

void relayS1On() {
digitalWrite(relayS1Pin, LOW); // turn on relay with voltage HIGH
}

void relayS1Off() {
digitalWrite(relayS1Pin, HIGH); // turn off relay with voltage HIGH
}

kakopappa, 2017-01-16:
I like to recommend to:

1. Make a simple test app to test the relay
2. Use the multi-meter or LED to check whether digitalWrite really works.
3. Check the power supply. make sure adruino is getting enough power to turn on the relay.

dsaleb, 2017-01-16:
I solved the problem.

On my relays High is off and Low is on. 

void S1On() {
    Serial.print("Switch 1 on ...");
    digitalWrite(relayS1Pin, LOW);
}

void S1Off() {
    Serial.print("Switch 1 off ...");
    digitalWrite(relayS1Pin, HIGH);
}

void S2On() {
    Serial.print("Switch 2 on ...");
    digitalWrite(relayS2Pin, LOW);
}

void S2Off() {
    Serial.print("Switch 2 off ...");
    digitalWrite(relayS2Pin, HIGH);
}

dsaleb, 2017-01-16:
I'm working on another project and I might need your help @joeman2116 @kakopappa 

Now I'm trying to add the following to this code

*web access to turn on/off from a browser
*using http widget mobile app to create on/off on screen shortcuts 
*using Blynk to remotely turn the devices on/off 

Would you guys be interested to contribute to this project?

DrexProjects, 2017-12-10:
I Re-worked the code a little to set relays to low at startup. It is as follows:

#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// Modified by DrexProjects

const int relayS1Pin = D1;
const int relayS2Pin = D2;
const int relayS3Pin = D3;
const int relayS4Pin = D5;

// prototypes
boolean connectWifi();

//on/off callbacks
void relayS1On();
void relayS1Off();
void relayS2On();
void relayS2Off();
void relayS3On();
void relayS3Off();
void relayS4On();
void relayS4Off();

// Change this before you flash
const char* ssid = "dlink"; // Replace with your router name
const char* password = "********";  // Add your password here

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *relayS1 = NULL;
Switch *relayS2 = NULL;
Switch *relayS3 = NULL;
Switch *relayS4 = NULL;

void setup()
{
Serial.begin(9600);

// Initialise wifi connection
wifiConnected = connectWifi();

if(wifiConnected){
upnpBroadcastResponder.beginUdpMulticast();

pinMode(relayS1Pin, OUTPUT);
pinMode(relayS2Pin, OUTPUT);
pinMode(relayS3Pin, OUTPUT);
pinMode(relayS4Pin, OUTPUT);

  digitalWrite(relayS1Pin, HIGH);       // sets the digital pin 1 on
  digitalWrite(relayS2Pin, HIGH);       // sets the digital pin 2 on
  digitalWrite(relayS3Pin, HIGH);       // sets the digital pin 3 on
  digitalWrite(relayS4Pin, HIGH);       // sets the digital pin 4 on
  
// Define your switches here. Max 14
// Format: Alexa invocation name, local port no, on callback, off callback
relayS1 = new Switch("relay one", 80, relayS1On, relayS1Off);
relayS2 = new Switch("relay two", 81, relayS2On, relayS2Off);
relayS3 = new Switch("relay three", 82, relayS3On, relayS3Off);
relayS4 = new Switch("relay four", 83, relayS4On, relayS4Off);

Serial.println("Adding switches upnp broadcast responder");
upnpBroadcastResponder.addDevice(*relayS1);
upnpBroadcastResponder.addDevice(*relayS2);
upnpBroadcastResponder.addDevice(*relayS3);
upnpBroadcastResponder.addDevice(*relayS4);
}
}

void loop()
{
if(wifiConnected){
upnpBroadcastResponder.serverLoop();

relayS1->serverLoop();
relayS2->serverLoop();
relayS3->serverLoop();
relayS4->serverLoop();
}
}

void relayS1On() {
digitalWrite(relayS1Pin, LOW); // turn on relay with voltage HIGH
}

void relayS1Off() {
digitalWrite(relayS1Pin, HIGH); // turn off relay with voltage HIGH
}

void relayS2On() {
digitalWrite(relayS2Pin, LOW); // turn on relay with voltage HIGH
}

void relayS2Off() {
digitalWrite(relayS2Pin, HIGH); // turn off relay with voltage HIGH
}

void relayS3On() {
digitalWrite(relayS3Pin, LOW); // turn on relay with voltage HIGH
}

void relayS3Off() {
digitalWrite(relayS3Pin, HIGH); // turn off relay with voltage HIGH
}
void relayS4On() {
digitalWrite(relayS4Pin, LOW); // turn on relay with voltage HIGH
}

void relayS4Off() {
digitalWrite(relayS4Pin, HIGH); // turn off relay with voltage HIGH
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
boolean state = true;
int i = 0;

WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
Serial.println("");
Serial.println("Connecting to WiFi");

// Wait for connection
Serial.print("Connecting ...");
while (WiFi.status() != WL_CONNECTED) {
delay(500);
Serial.print(".");
if (i > 10){
state = false;
break;
}
i++;
}

if (state){
Serial.println("");
Serial.print("Connected to ");
Serial.println(ssid);
Serial.print("IP address: ");
Serial.println(WiFi.localIP());
}
else {
Serial.println("");
Serial.println("Connection failed.");
}

return state;
}