kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #8: randomly disconnecting and reconnecting from wifi

xal3xhx, 2017-01-19:
great work with this, i use it almost everyday but the random blue flashing light is getting annoying. when left on for a min or two it spams the serial port with a bunch of random stuff then acts as if it reconnected to the wifi. i'm not sure if its really disconnecting and reconnecting or if its just a bug. it seams like the longer i leave it powered the more it spams.

```
Exception (0):
epc1=0x40227bc0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  60000600 3ffeefa8 3ffe9ea0 3ffe9ea0  
3fffff80:  4021d796 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeefe8 00000000 3fff3b6c 4021d3bb  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<
c?ÇrSöfjø
Connecting to WiFi
Connecting ...
Exception (0):
epc1=0x402091a4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40209111 3fffdab0 00000000 00000000  
3fffff80:  3fff1b74 40208b3a 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe9e20 00000000 40206f4f  
3fffffa0:  3ffe9e20 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_GrSö2jø
Connecting to WiFi
Connecting ...
Exception (0):
epc1=0x402091a7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40209111 3fffdab0 00000000 00000000  
3fffff80:  3fff1b74 40208b3a 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe9e20 00000000 40206f4f  
3fffffa0:  3ffe9e20 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_Ç"Söfjø
Connecting to WiFi
Connecting ...
Exception (28):
epc1=0x40209218 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000024 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40209111 3fffdab0 00000000 00000000  
3fffff80:  3fff1b74 40208b3a 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe9e20 00000004 40206f4f  
3fffffa0:  3ffe9e20 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_ÇrSö&jø
Connecting to WiFi
Connecting ...
Exception (9):
epc1=0x402091a4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000085 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40209111 3fffdab0 00000000 00000000  
3fffff80:  3fff1b74 40208b3a 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe9e20 00000000 40206f4f  
3fffffa0:  3ffe9e20 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_ÇrSö&jø
Connecting to WiFi
Connecting ...
Exception (0):
epc1=0x402091a7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000085 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40209111 3fffdab0 00000000 00000000  
3fffff80:  3fff1b74 40208b3a 3fffdab0 00000000  
3fffff90:  3fffdcc0 3ffe9e20 00000000 40206f4f  
3fffffa0:  3ffe9e20 40000f49 3fffdab0 40000f49  
<<<stack<<<
c_ÇrSöfjø
Connecting to WiFi
Connecting .........
Connected to Home
```

xal3xhx, 2017-01-19:
so i tried re flashing the firmware and it did not work but using another esp seams to work not sure what the problem is.

closed 2017-01-19 by xal3xhx

tvixen, 2017-01-20:
Make sure to solder a 100uf over the supply.