kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #9: Can you add eeprom- so that the switch state in the memory of ESP incase of blackout

sayemahmed, 2017-01-23:
Thanks for this nice work. It helps me a lot. I was suing some RF based system in my home. Now I want to replace with Alexa. In my earlier system I used EEPROM to keep switch state stored in the memory. I was trying to add codes to add EEPROM in your code. But it generates WDT RESET! ESP is continuously restarting. Please help with this.

sayemahmed, 2017-01-23:
My code here,
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <EEPROM.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// prototypes
boolean connectWifi();

//on/off callbacks 
void officeLightsOn();
void officeLightsOff();
void kitchenLightsOn();
void kitchenLightsOff();

// Change this before you flash
const char* ssid = "Tenda_55A098";
const char* password = "Gaibandha*72";

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

// Define relay
const int relayS1Pin = D1;
const int relayS2Pin = D2;
const int relayS3Pin = D3;
const int relayS4Pin = D4;

// Switch
Switch *s1 = NULL;
Switch *s2 = NULL;
Switch *s3 = NULL;
Switch *s4 = NULL;
//

void setup()
{
  Serial.begin(115200);
   
  // Initialise wifi connection
  wifiConnected = connectWifi();
  
  if(wifiConnected){
    upnpBroadcastResponder.beginUdpMulticast();
    
    // Define your switches here. Max 14
    // Format: Alexa invocation name, local port no, on callback, off callback
    //office = new Switch("office lights", 80, officeLightsOn, officeLightsOff);
    //kitchen = new Switch("kitchen lights", 81, kitchenLightsOn, kitchenLightsOff);
      s1 = new Switch("office", 80, s1On, s1Off);
      s2 = new Switch("kitchen", 81, s2On, s2Off);
      s3 = new Switch("fan", 82, s3On, s3Off);
      s4 = new Switch("tv", 83, s4On, s4Off);

//////////////////////////////////
// Inside void setup() 
 pinMode(relayS1Pin, OUTPUT);
 pinMode(relayS2Pin, OUTPUT);
 pinMode(relayS3Pin, OUTPUT);
 pinMode(relayS4Pin, OUTPUT);

upnpBroadcastResponder.addDevice(*s1);
upnpBroadcastResponder.addDevice(*s2);
upnpBroadcastResponder.addDevice(*s3);
upnpBroadcastResponder.addDevice(*s4);

//////////////////////////////////
    Serial.println("Adding switches upnp broadcast responder");
    upnpBroadcastResponder.addDevice(*s1);
    upnpBroadcastResponder.addDevice(*s2);
    upnpBroadcastResponder.addDevice(*s3);
    upnpBroadcastResponder.addDevice(*s4);        

//EPROM
   ESP.wdtDisable();
   ESP.wdtEnable(WDTO_8S);
   EEPROM.begin(4);
  //-----------------------EEPROM CH1------------------------------
  if (EEPROM.read(0) == 1)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS1Pin, HIGH) ; 
  }
  if (EEPROM.read(0) == 0)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS1Pin, LOW) ;
  }
 //-----------------------EEPROM CH2------------------------------
  if (EEPROM.read(1) == 1)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS2Pin, HIGH) ;
  }
  if (EEPROM.read(1) == 0)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS2Pin, LOW) ;
  }
 //-----------------------EEPROM CH3------------------------------ 
  if (EEPROM.read(2) == 1)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS3Pin, HIGH) ; 
  }
  if (EEPROM.read(2) == 0)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS3Pin, LOW) ; 
  }
   //-----------------------EEPROM CH4------------------------------ 
  if (EEPROM.read(3) == 1)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS4Pin, HIGH) ; 
  }
  if (EEPROM.read(3) == 0)
  {  // switch is pressed - pullup keeps pin high normally
       digitalWrite(relayS4Pin, LOW) ; 
  }
//-------------------------EEPROM END------------------------------
    
  }
}
 
void loop()
{
   ESP.wdtFeed();
   if(wifiConnected){
      upnpBroadcastResponder.serverLoop();
      
      s1->serverLoop();
      s2->serverLoop();
      s3->serverLoop();
      s4->serverLoop();
   }
}

void s1On() {
  digitalWrite(relayS1Pin, HIGH); // turn on relay with voltage HIGH 
  EEPROM.write(0, 1);
  Serial.print("Switch 1 turn on ...");
}

void s1Off() {
  digitalWrite(relayS1Pin, LOW); // turn off relay with voltage HIGH 
  EEPROM.write(0, 0);
  Serial.print("Switch 1 turn off ...");
}

void s2On() {
  digitalWrite(relayS2Pin, HIGH); // turn on relay with voltage HIGH 
  EEPROM.write(1, 1);
  Serial.print("Switch 2 turn on ...");
}

void s2Off() {
  digitalWrite(relayS2Pin, LOW); // turn off relay with voltage HIGH 
  EEPROM.write(1, 0);
  Serial.print("Switch 2 turn off ...");
}

void s3On() {
  digitalWrite(relayS3Pin, HIGH); // turn on relay with voltage HIGH 
  EEPROM.write(2, 1);
  Serial.print("Switch 3 turn on ...");
}

void s3Off() {
  digitalWrite(relayS3Pin, LOW); // turn off relay with voltage HIGH 
  EEPROM.write(2, 0);
  Serial.print("Switch 3 turn off ...");
}

void s4On() {
  digitalWrite(relayS4Pin, HIGH); // turn on relay with voltage HIGH 
  EEPROM.write(3, 1);
  Serial.print("Switch 4 turn on ...");
}

void s4Off() {
  digitalWrite(relayS4Pin, LOW); // turn off relay with voltage HIGH 
  EEPROM.write(3, 0);
  Serial.print("Switch 4 turn off ...");
}

// connect to wifi – returns true if successful or false if not
boolean connectWifi(){
  boolean state = true;
  int i = 0;
  
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");
  Serial.println("Connecting to WiFi");

  // Wait for connection
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }
  
  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed.");
  }
  
  return state;
}

joeman2116, 2017-01-23:
Looks good.  Adding the eeprom support brings the code up another notch.!!
Have not tried it yet but will give it a shot in a few days..

Thanks!
Joe

kakopappa, 2017-01-24:
Does it work fine without your modifications ? 

If Yes, try switching to FS.h
https://blog.squix.org/2015/08/esp8266arduino-playing-around-with.html

You can always use the EspExceptionDecoder to check where is the exception coming from
 https://github.com/me-no-dev/EspExceptionDecoder

sayemahmed, 2017-01-24:
It is not working with my current Node mcu. It is may be broken. I will check with a fresh Node Mcu next Saturday.

kakopappa, 2017-01-24:
just came across this 

https://espressif.com/sites/default/files/documentation/esp8266_reset_causes_and_common_fatal_exception_causes_en.pdf

sayemahmed, 2017-01-24:
Many thanks for this document.

sayemahmed, 2017-01-28:
The code above does not working! Please help...

cliffspr, 2018-01-31:
Hi, I have had great success your library, and just wondered how I can add a manual switch and update alexa with the change. I have done it successfully with your single switch but finding the more library based multy switch more dificault to get my head around. Any help would be much appreciated. Regards, Cliff

parkash8266, 2018-03-22:
hello sir 
this is a best for home Automaton project .
when i load the code in my nodmcu then the massage show 
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
i have selected proper board ,upload speed 115200, 80mhz
what is solution
my mail. parmatma8819@gmail.com