kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #13: Exception (9): crashing

Svengali86, 2017-03-04:
Anyone else having trouble with this crashing?

I'm using a nodemcu board, and randomly it will just crash and dump a load of stuff in the serial monitor.

Tried multiple boards with the same outcome.

I can't seem to install the stack decoder on my arduino IDE running windows, any help would be great.

Best regards

kakopappa, 2017-03-08:
Can you remove all Serial.Println check please.

Svengali86, 2017-03-08:
Hi, thanks for your response.

Still no change unfortunately.

It works, but will randomly drop out, after 10 - 60s

2 "devices" switching 2 relays.

Your code for single switch works flawlessly - thank you so much.

kakopappa, 2017-03-10:
What is your ESP module ?

Have you tried looking at the crash log?
https://github.com/me-no-dev/EspExceptionDecoder

Svengali86, 2017-03-10:
Hi, 

Using these, also tried a wemos d1 mini

[http://www.electrodragon.com/product/nodemcu-lua-amica-r2-esp8266-wifi-board/](url)

Managed to get the decoder to work here is some of it;

0x4021d535: system_get_os_print at ?? line ?
0x4021d6e7: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x4010164a: pp_post at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x401024ec: trc_NeedRTS at ?? line ?
0x402348f0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x4021c493: user_uart_wait_tx_fifo_empty at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x4010164a: pp_post at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40103749: lmacIsIdle at ?? line ?
0x40103749: lmacIsIdle at ?? line ?
0x4020cdd3: pm_post at ?? line ?
0x4021d53e: system_get_os_print at ?? line ?
0x4010164a: pp_post at ?? line ?
0x4021d6e7: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x4010164a: pp_post at ?? line ?
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40104927: lmacRxDone at ?? line ?
0x401024ec: trc_NeedRTS at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/Switch.cpp line 36
0x402348f0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/Switch.cpp line 36
0x4021c493: user_uart_wait_tx_fifo_empty at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/Switch.cpp line 36
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/Switch.cpp line 36
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/Switch.cpp line 36
0x4020299c: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 64
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x402064c8: String::changeBuffer(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40206517: String::reserve(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4020646c: String::~String() at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40202885: UpnpBroadcastResponder::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/UpnpBroadcastResponder.cpp line 84
0x40202994: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 63
0x40206c78: loop_wrapper at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

Svengali86, 2017-03-10:
Also

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 5 results
0x40204abf: ESP8266WebServer::handleClient() at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\src\detail/RequestHandlersImpl.h line 53
0x40202885: UpnpBroadcastResponder::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_763657\sketch/UpnpBroadcastResponder.cpp line 84
0x4020299c: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 64
0x40206c78: loop_wrapper at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

kakopappa, 2017-03-14:
I think this is related to power supply . Can you try using a external power supply other than FTI

Svengali86, 2017-03-20:
Hi,

Sorry I've been away a while.

I'm using a samsung phone charge 5v 2a, 10cm usb cable. 

What would you suggest to use, as was hoping I could use these for the finished project like I have with the single switches.

With best regards

Svengali86, 2017-03-21:
[image]

Tried this still same

Svengali86, 2017-03-24:
So I bought a adafruit feather huzzah esp8266.

Still the same story. here are some logs:

Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 32 results
0x401050f1: ets_timer_disarm at ?? line ?
0x4020cddb: pm_post at ?? line ?
0x4021d546: system_get_os_print at ?? line ?
0x4022675c: tcp_create_segment at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 166
0x4021d6ef: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x4020342e: ClientContext::write(char const*, unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40234900: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x4021c49b: user_uart_wait_tx_fifo_empty at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x402029a3: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 65
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x402064d0: String::changeBuffer(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4020651f: String::reserve(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40206474: String::~String() at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40202885: UpnpBroadcastResponder::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/UpnpBroadcastResponder.cpp line 84
0x402010ae: delay at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 53
0x4020299a: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 64
0x40206c80: loop_wrapper at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

Svengali86, 2017-03-24:
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 60 results
0x401050f1: ets_timer_disarm at ?? line ?
0x4021d546: system_get_os_print at ?? line ?
0x4021d6ef: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x401050f1: ets_timer_disarm at ?? line ?
0x40234900: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x4021c49b: user_uart_wait_tx_fifo_empty at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40105139: ets_timer_disarm at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40101793: ppCalFrameTimes at ?? line ?
0x40228504: etharp_output at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c line 995
0x401026c4: wDev_ProcessFiq at ?? line ?
0x4022944c: ip_output_if_opt at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c line 780
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4020cddb: pm_post at ?? line ?
0x4021d546: system_get_os_print at ?? line ?
0x4022675c: tcp_create_segment at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/tcp_out.c line 166
0x4021d6ef: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x4020342e: ClientContext::write(char const*, unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40234900: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x4021c49b: user_uart_wait_tx_fifo_empty at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x40202047: Switch::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/Switch.cpp line 36
0x402029a3: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 65
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x402064d0: String::changeBuffer(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4020651f: String::reserve(unsigned int) at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x4010020c: _umm_free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x40206474: String::~String() at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/WString.cpp line 519
0x40202885: UpnpBroadcastResponder::serverLoop() at C:\Users\Dan\AppData\Local\Temp\arduino_build_921397\sketch/UpnpBroadcastResponder.cpp line 84
0x402010ae: delay at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 53
0x4020299a: loop at C:\Users\Dan\Downloads\arduino-esp8266-alexa-multiple-wemo-switch-master3\arduino-esp8266-alexa-multiple-wemo-switch-master\wemos/wemos.ino line 64
0x40206c80: loop_wrapper at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109

Svengali86, 2017-03-24:
Decoding 18 results
0x4021d53d: system_get_os_print at ?? line ?
0x4021d6ef: system_restart_local at ?? line ?
0x40201981: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40106afc: delayMicroseconds at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_wiring.c line 74
0x40201997: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x40201994: __wrap_system_restart_local at C:\Users\Dan\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_postmortem.c line 190
0x401050f1: ets_timer_disarm at ?? line ?
0x4010164a: pp_post at ?? line ?
0x40234900: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x4021c49b: user_uart_wait_tx_fifo_empty at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x40104c27: wdt_feed at ?? line ?
0x40104c08: wdt_feed at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?
0x401050f1: ets_timer_disarm at ?? line ?

PhilW-WBC, 2017-04-19:
Just a note here - I also get the crashes with the Adafruit Huzzah, but I do not get them with a SparkFun 8266 Thing (Dev or Standard.)  So it does point to the FTDI boards not providing enough power through the programming interface.

Svengali86, 2017-04-20:
Thanks, I've used 3 different boards with the same outcome. Even the wemos mini as developed on. I have given up as I don't want to waste any more money on boards

kakopappa, 2017-07-27:
try https://bitbucket.org/xoseperez/fauxmoesp if this library does not work

pridiol, 2018-01-07:
Had the same issue at home, where I have lots of devices speaking unpin (2 echos, 3 firetvs, some sonos speakers, web radios, sonoff bridge, ...)

At the office, with only one echo dot, it worked fine.

I tried nearly everything. Then I saw the line "char packetBuffer[512];" in UpnpBroadcastResponder.cpp. 

I raised this buffer to 1024 bytes.
Now everything works perfectly. No crashes any more ;-)

kakopappa, 2018-01-09:
fix added to master

closed 2018-01-09 by kakopappa

IraSch, 2018-02-20:
Hi, I'm having a problem with compiling. At first I downloaded your wemos project, compiled it, ran it and all was well. Alexa interface working just fine. So I then went in and added 3 lines: initialize the D0 port, and set D0 HIGH or LOW depending on whether Kitchen Switch was on or off. 

Wouldn't compile. Got error:   'Redefinition of const char* ssid'

So I went back to the original wemos.ino file and get the same error. Rebooted, still same error. Using Arduino version 1.8.5

Any idea why I'd get this compiler error message?

Thanks, Ira

sachin-shrirame, 2018-07-17:
Getting Below Error, Can you please tell me what happaned wrong 

UpnpBroadcastResponder.cpp:70:22: warning: character constant too long for its type [enabled by default]
   if(request.indexOf('M-SEARCH') >= 0) {
                      ^
UpnpBroadcastResponder.cpp: In member function 'void UpnpBroadcastResponder::serverLoop()':
UpnpBroadcastResponder.cpp:70:32: warning: large integer implicitly truncated to unsigned type [-Woverflow]
   if(request.indexOf('M-SEARCH') >= 0) {
                                ^

kakopappa, 2018-07-18:
Looks like warnings not errors.