kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #14: Where are the Wemos output pins defined?

dougstrickland, 2017-03-07:
I see the port numbers in the wemos.ino file, but I don't see where the specific GPIO pins are defined.  Where do I find the D1..outputs defined?

kakopappa, 2017-03-08:
https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L49-L61

kakopappa, 2017-03-08:
Here is the pin layout
http://community.blynk.cc/uploads/default/original/2X/4/4f9e2245bf4f6698e10530b9060595c893bf49a2.png

in Arduino IDE, If you select Tools ->Board-> Generic, D1 is not available, You have to copy it  from  https://github.com/esp8266/Arduino/blob/master/variants/d1_mini/pins_arduino.h#L49-L61

 If you select WeMos D1 it should work fine.

dougstrickland, 2017-03-08:
You are so helpful. Thanks.

closed 2017-03-08 by dougstrickland