kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #16: What do these errors mean and how do I fix them

zachJsmith, 2017-03-30:
Arduino: 1.8.2 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, QIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

Archiving built core (caching) in: /var/folders/0s/tcclhfjs2p92xhrr0yd0ryx40000gn/T/arduino_cache_110302/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_qio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____c705db4fc699f6463c021fb2c3716b5a.a
Sketch uses 249701 bytes (57%) of program storage space. Maximum is 434160 bytes.
Global variables use 37500 bytes (45%) of dynamic memory, leaving 44420 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kakopappa, 2017-03-30:
Try changing the reset method to  mcu in Arduino IDE

zachJsmith, 2017-03-30:
i could ony find node mcu but, it still says this:
Arduino: 1.8.2 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 40MHz, QIO, 115200, 512K (64K SPIFFS), nodemcu, Disabled, None"

Build options changed, rebuilding all
Archiving built core (caching) in: /var/folders/0s/tcclhfjs2p92xhrr0yd0ryx40000gn/T/arduino_cache_736694/core/core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_qio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_nodemcu,Debug_Disabled,DebugLevel_None_____c705db4fc699f6463c021fb2c3716b5a.a
Sketch uses 249701 bytes (57%) of program storage space. Maximum is 434160 bytes.
Global variables use 37500 bytes (45%) of dynamic memory, leaving 44420 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kakopappa, 2017-03-30:
Do you have a flash button?

Try holding the flash button before flashing start

zachJsmith, 2017-03-30:
like a physical button or in the software

kakopappa, 2017-03-30:
Can you send me a link to your module?

zachJsmith, 2017-03-30:
https://www.amazon.com/gp/product/B012YUANZK/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

zachJsmith, 2017-03-30:
the realy board was the link node r4

kakopappa, 2017-03-31:
This is a Ftdi Usb to Ttl Serial Adapter .. You need https://www.aliexpress.com/store/product/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/1331105_32529101036.html

hojatsadat, 2017-05-09:
I have the same issue but I'm using this one https://www.google.dk/search?q=esp8266&client=ubuntu&hs=SqO&channel=fs&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiG2pyvwuLTAhXnZpoKHes9CvoQ_AUICigB&biw=1920&bih=1110#imgrc=Z2zdlsmWCMB6rM:

How Can I solve this ?

mfatihkoc, 2019-03-21:
Hi all, What is the problem with this code?

Archiving built core (caching) in: C:\Users\Fatih\AppData\Local\Temp\arduino_cache_97831\core\core_arduino_avr_nano_cpu_atmega328_0c812875ac70eb4a9b385d8fb077f54c.a

kakopappa, 2019-03-21:
have you selected the correct dev board ? looks like it's compiling for atmega328.

mfatihkoc, 2019-03-21:
I am using A4988 stepper motor module.

mfatihkoc, 2019-03-21:
I am trying to see whether my nema 17 stepper motor works or? 
I have nano arduino, a4988 motor driver module and nema17.
do you have any works about it?

mfatihkoc, 2019-03-21:
attention : MPU6050 library ''. 'Uncategorized" know is a fault I have

mfatihkoc, 2019-03-21:
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : arduino
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 2
         Firmware Version: 1.16
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\Fatih\AppData\Local\Temp\arduino_build_180802/steppmotor1.ino.hex"
avrdude: writing flash (1048 bytes):

Writing | ################################################## | 100% 0.33s

avrdude: 1048 bytes of flash written
avrdude: verifying flash memory against C:\Users\Fatih\AppData\Local\Temp\arduino_build_180802/steppmotor1.ino.hex:
avrdude: load data flash data from input file C:\Users\Fatih\AppData\Local\Temp\arduino_build_180802/steppmotor1.ino.hex:
avrdude: input file C:\Users\Fatih\AppData\Local\Temp\arduino_build_180802/steppmotor1.ino.hex contains 1048 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.25s

avrdude: verifying ...
avrdude: 1048 bytes of flash verified

avrdude done.  Thank you.