kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #19: Exception (9)

bryanberger, 2017-07-20:
Got this working on my NodeMCU v2 ESP 12-E module but every so often I get an Exception and the NodeMCU crashes.

here is the stack trace
```
Exception (9):
epc1=0x40202047 epc2=0x00000000 epc3=0x00000000 excvaddr=0x73626577 depc=0x00000000

ctx: cont 
sp: 3fff07b0 end: 3fff0990 offset: 01a0

>>>stack>>>
3fff0950:  3fffdad0 00000000 3ffef7b0 3ffef968  
3fff0960:  3fffdad0 00000000 3ffef7b0 40202aec  
3fff0970:  3fffdad0 00000000 3ffef960 40206c10  
3fff0980:  feefeffe feefeffe 3ffef970 40100718  
<<<stack<<<
⸮⸮P9⸮⸮
```

Sometimes it initializes sometimes it doesn't. Any idea what this error means?

kakopappa, 2017-07-20:
Not sure what's causing it. you try this https://bitbucket.org/xoseperez/fauxmoesp . does it work?

bryanberger, 2017-07-20:
@kakopappa cloned [fauxmoesp](https://bitbucket.org/xoseperez/fauxmoesp) and it seems to be working, no hiccups yet, thanks for the suggestion!