kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #21: Wemos multi devices

Hasci5, 2017-10-17:
Hi everyone I call Hasci and thanks for the great program is great. I use a Nodemcu 1.0 and I only have a problem: up to the fourth alexa device recognizes it as soon as I insert the 5th and 6th and 7th so until the alexa slide recognizes only the last 4 devices (example: if I use ports 80 -81-82-83-84-90-91) alexa only recognizes 83-84-90-91 always and only the last 4 devices. I wonder if it happened to you. thank you in advance.

riban-bw, 2017-11-19:
I am also suffering this issue. I can't figure it out yet. All 8 Switches are created and server ports are open but the same (last) 4 responses are sent twice to the Alexa scan.

riban-bw, 2017-11-19:
I found that the code could not bind to port 87 (which may suggest some other code has this port open - should check for security hole). Changed my port numbers to 8081 - 8088 and got same results (last 4 switches sent twice during discovery) but am able to connect to each port and see correct response to /setup.xml so looks like an issue in the discovery code....

riban-bw, 2017-11-19:
Added more debug and I see that this library appears to be sending the correct response to discovery so maybe the Belkin skill is limited to 4 switches per host. If so - there isn't much we can do about this other than limit this code to 4 switches (or possibly add more IP addresses but I don't know if the network library supports that).

riban-bw, 2017-11-25:
I have changed the code quite a lot and now see that the Alexa skill is able to handle more than 4 connections so I suspect the issue I saw was to do with message timing (you sometimes need to scan for devices several times) and some kind of reference overlap???? Anyway - ignore last comment about not handling more than 4 devices on one IP address. I have found a limit due to the quantity of webservers and hence the memory usage (probably). If I instantiate more that 12 switches the program crashes but even with 12 it will also crash when http traffic begins. 8 or 9 works - 10 gets a bit temperamental.

juanjocastillo, 2017-11-29:
Hi!! Where can i find this new code? In this repository says it was modified one year ago...