kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #22: Alexa SmartHome skill - not finding

sciencedude100, 2017-11-02:
Installed on my NodeMCU, went through Alexa app to add... nothing.
So, turns out you need the smart home skill for the devices you want to add.
The WeMo skill requires you to enter your wireless SSID and the WeMo device's MAC address.
This breaks the compatibility with these wemo emulator projects unfortunately.

So, has anyone gotten past this? Is there a generic smarthome skill I'm unaware of? 

P.S. I made sure all devices were on the same network and I even reset my Echo Dot. Nothing :/

dougstrickland, 2017-11-03:
Have you tried coding your ssid in your script?  
like:

const char* ssid = "My_Network";
const char* password = "XXXXXXXXXXXXXXXX";

sciencedude100, 2017-11-03:
I set the SSID and WPA2 password, I was able to ping it from my PC as well as my phone from the same network that the Echo is on

Braandn, 2017-11-04:
Same problem

MacSass, 2017-11-05:
Same problem here - but only after my Alexa updated to the latest firmware.
Before I could discover the ESP emulated devices with no problem - you could monitor the answer to the Alexa search request in the serial console ...
Now, when I do a search, the ESP does not seem to "see" any broadcast request and does not respond - thus Alexa does not show any devices.
I will try with the WeMo skill, but if that required a MAC address and it only works this way it would mean we would need "false" MAC adresses???
As said - until lately no skill was needed and Alexa found the emulated Wemos devices on the ESP8266 with no problem ...

MacSass, 2017-11-05:
Update:
Using my "old" Amazon Echo (1st generation) the devices are discovered with no skill required (no wemos skill activated). Devices can be controlled. The Echo FW version is 591448720.
The second generation Echo does not seem to be able to control or discover the devices with the latest Fw that was released today.
Looks like Amazon is really messing up big time with the second generation Echo (sound is way worse than with the old one - now this).
Let´s see if Amazon fixes that or if the kill my old Echo for this by updating it´s FW as well ...

sciencedude100, 2017-11-08:
maybe someone just needs to make a smart home skill for the ESP8266. I might give it a shot, but no guarantees.

arnowillig, 2017-11-24:
@MacSass Same here, 2nd gen. Echo does not finde the devices, Echo FW version is 592452720.
However, the echo DOES send HTTP requests (GET to /setup.xml and /eventservice.xml and POST to /upnp/control/basicevent1).
POST body is this:
`<?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>`

I think the echo tries to read the current state of the emulated WeMo switch, but then fails to enumerate the device. :-(

Anyone a solution yet?
akw

kakopappa, 2017-11-25:
hello @MacSass @arnowillig @sciencedude100 @Braandn 

Thank you for your input guys 

I have added GetBinaryState support to single switch version. Can you guys test it and let me know. I have an Eco at home. It does not send GetBinaryState 

https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066

ma7gdp, 2017-11-26:
I just tried the code you have on a "WemosD1 mini" and the echo dot 2nd gen is still reports "No new devices found".  The serial on the WemosD1 is outputting the usual though: 

Sending response to 192.168.1.74
Port : 50000
Response sent !
192.168.1.74, port 50000
192.168.1.74, port 50000
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>box</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e61957d9</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>box</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e61957d9</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

RootCNC, 2017-12-01:
Same, doesn't work with Sonos One with Alexa built in- unable to discover devices =[ great project and keep up the good work! I will be following it!

kakopappa, 2017-12-02:
@RootCNC can you try https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066

tommyfranke, 2017-12-03:
@kakopappa, thank's for the great work.
The above code works wonderful on my new Amazon Dot V2 with actual Firmware 595459620.
My Nodemcu was found at once, you don't need the Wemo-Skill!!!
Switching works fine.

Btw: It seems to me that the "eventservice.xml" never get called by Echo Dot, is it not needed?

ZoMiGo, 2017-12-06:
hi i have  Amazon Dot V2 with actual Firmware 595459620 my problem is alexa dont found any devices.
i use  https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066

arnowillig, 2017-12-06:
hi, same here. I got an Amazon Echo gen.2.
Even with the original WeMo plug the Amazon Echo does not find the hardware.
However, I can see the device is queried. Even an "GetDeviceState" upnp
POST message is sent by the Echo hardware.
Strange, hm? I suppose this is a bug in the firmware.
My Amazon Echo gen.1 works beautifully with the original WeMo plug as well
with a simulated device.

Ciao, Arno

RootCNC, 2017-12-08:
@kakopappa I've just ran the software now and unfortunately it doesn't seem to work. =[ 

I have gone into the Alexa app to see the version number and its not listed where amazon says it should, I was hoping this might help debug the problem. 

I can see in the serial terminal Alexa is sending something to the ESP8266 but still doesnt discover any devices. =[ 

The Sonos one with Alexa does with the Wemo insight version perfectly so not too sure.

sciencedude100, 2017-12-17:
This still isn't found on my Alexa V2 (Firmware: 597464620)

MacSass, 2017-12-17:
Thanks for your work on this Kakopappa - I will try this latest version during the week, although I do not have too much hope looking at past responses.
I could provide some network trace if that helps if you let me know what you would need ...

MacSass, 2017-12-18:
Hello, I have tried with the new script found here: 
[https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066]
and can confirm that no devices are found with that script with the new generation Alexa devices (tried Echo v2 and Echo Plus).
Both - the old and the new script - do not show any reaction at all when I do a "discover" in the Alexa app if there is only a new generation Alexa doing the search.
Interesting fact: 
As soon as I have one "old" Alexa device (Echo Dot in my case) in the same network then both scripts do find devices with no problem.
It looks like the new devices send out a different broadcast string, as I never see a response from the ESP with the new devices. I would have expected that this:

    if(request.indexOf('M-SEARCH') > 0) {
            if(request.indexOf("urn:Belkin:device:**") > 0) {
                Serial.println("Responding to search request ...");
                respondToSearch();
            }
        }

get´s triggered - but it does not with the new device (ip ....28 in attached logs), only if an old device is in the network as well the ESP responds to the broadcast of the old device (ip ...24 in attached logs)

I have attached the output of the script for V2 device only and for combination of V2+V1 Alexa device.

I think the broadcast of the generations is different and the listening ESP never get´s triggered by the new device. Thus it seems to be needed to modify the trigger criteria?
If I find the time I will see if I can do a network trace to check what the difference in query broadcast of the two generations is, that should clarify things.

Regards - MacSass

[New script - Echo Plus and Echo Dot.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1569421/New.script.-.Echo.Plus.and.Echo.Dot.txt)
[New script - Echo Plus only.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1569422/New.script.-.Echo.Plus.only.txt)

MacSass, 2017-12-18:
Hi,
I did a network trace and think I´m getting a little bit clearer on the issue:
While old devices send a SSDP broadcast containing "urn:Belkin:device:**" (which the ESP is looking for) the new devices send a SSDP broadcast with "upnp:rootdevice". Thus the ESP does never respond to the broadcast of the new Alexa devices.

When I do change the trigger that ESP is looking for to:
`
if(request.indexOf('M-SEARCH') > 0) {
        if(request.indexOf("**upnp:rootdevice**") > 0) {
            Serial.println("Responding to search request ...");
            respondToSearch();
        }
    }
`

and also change the response string, then I can get the new device to discover the ESP!

I did an initial test and the script then seems to work fine for me. I have attached my modified script below and would appreciate any feedback, as I might not really fully understand what I´m doing :-) 

[irdevkit-macsass.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1569627/irdevkit-macsass.txt)

Regards - MacSass

sciencedude100, 2017-12-19:
[**It Works!**](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch)
[Arduino Log](https://gist.github.com/sciencedude100/20d231e9e423e0ca87efefd0bb251935)

I ended up having to reset my Echo Dot V2 ( hold Mute & Vol - ).
Make sure the echo and esp8266 are on the same wireless. I've had issues with them on the same network, but on different APs.

closed 2017-12-19 by sciencedude100

ma7gdp, 2017-12-20:
Is anyone else having an issue with getting this working? The serial output on my WemosD1mini shows activity when EchoDotV2 is discovering, but only ever shows "########## Responding to setup.xml ... ########" never the piece which @sciencedude100 shows in their Arduino log:  "########## Responding to eventservice.xml ... ########". It feels like it is almost there.

MacSass, 2017-12-20:
Hello ma7gdp,
yes - I´m still having issues. As long as I only have my Echo Plus (new gen) in the network, devices still do NOT get discovered. I see requests for UDP Belkin in the log and responses being sent, but devices do not show up.
Also I do NOT see responses like "######### Responding to setup.xml ... ########".

As much as I can say this is caused by the fact that the response that get´s send currently is having:
"USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n"
while the new gen devices expect a response of: 
"USN: uuid:" + persistent_uuid + "::upnp:rootdevice\r\n"

I will verify my assumption in a minute, but I´m pretty confident because as soon as I add a 1st gen device (echo dot) to the network the discovery works fine (because 1st gen device is happy with the response).

Doing some more testing right now ...

MacSass, 2017-12-20:
Hi again,
ok - I can confirm that this seems to be the root cause.
Changing 
"USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n"   in switch.cpp to
"USN: uuid:" + persistent_uuid + "::upnp:rootdevice\r\n"
does allow to discover devices with 2nd generation Echos. Also it still works if only a 1st gen Echo is in the network - so this seems to be a solution as much as I can see.

After changing a devices state I do see continious responses to 
###### Responding to  /upnp/control/basicevent1 ... ###
about every 1 second. Not sure if those requests really happen or if this is another issue that should be looked at in a different thread / issue.

ma7gdp, 2017-12-21:
Hey thanks @MacSass for taking a look. I only have Gen2 echo dot so cannot test the various combinations of scenarios with Gen1 etc. I can say though that even with the change to rootdevice in switch.cpp there are still no devices found by Alexa on the Gen2 dot.  The serial output I get is as below (.97 is the Gen2 and the WemosD1mini is on .98). Still just "Responding to setup.xml" nothing else. Looks like the Gen2 is expecting a different response to the setup.xml request (assuming that the setup.xml request is Gen2 asking WemosD1mini to identify itself so Gen2 can list it via Alexa app).

btw... Thanks for taking the time. If only I had a Gen1 ... I probably would not have even noticed things weren't quite as they should be.

> Udp multicast server started at 239.255.255.250:1900
> WebServer started on port: 80
> WebServer started on port: 81
> Adding switches upnp broadcast responder
> Adding switch : office lights index : 0
> Adding switch : kitchen lights index : 1
> Got UDP Belkin Request..
> 
> Sending response to 192.168.1.97
> Port : 50000
> Response sent !
>  ########## Responding to setup.xml ... ########
> 
> Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>office lights</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
> <UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e61957d9-80</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

MacSass, 2017-12-21:
Hi ma7gdp,
I can get devices discovered with both, my Echo Dot (current version - whatever that is) and my "new" Echo Plus, depending on what I put into the switch.cpp. But I still have some issues with devices switching if only one of them is on the network.
Currently I´m not clear if devices are "linked" to a certain Echo if they are only linked to the account. 
So I need to do more testing or additional network traces to get clarity on this.

So there definitely seems still something to be going on in what the devices expect as response for discovery and switching.
Fortunately the Christmas break is coming up and I should have some more time to look into this into more detail and do more testing.
I hope to have some more information / data in the upcoming days ... hopefully even before Christmas ...

ma7gdp, 2017-12-21:
Great. No more work this year for me so happy to help out with any testing or traces at my end. Tricky part is understanding whats in the echo code without the original/working devices to "sniff". Cheers again.

ma7gdp, 2017-12-22:
Picked up on your comment regarding the accounts. After lots of various testing I found issues with having a "household" Amazon account. Saying "Alexa switch accounts" and then "Alexa discover devices" did the trick.  Not only did the Echo find the devices but it also allowed me to trigger them and the serial output of the WemosD1mini showed the catch of the event. Bottom line .... I can now trigger a relay by saying "Alexa switch office lights on" etc.. Thanks all. To confirm I have 2x latest EchoDots only on my network.

jerrishjacob, 2017-12-25:
Hi ma7gdp, may I request you to post the modified code for esp8266 nodemcu. Really appreciate your efforts to solve this issue.

ioanvapi, 2017-12-28:
Hi. I have an Echo Dot 2nd gen. and loaded my nodemcu with code from main but it didn't work. I have started Wireshark and monitor the communication on UDP port 1900 and asked Alexa to discover devices. I got the upnp search message sent by my Echo and the message starts with "M-SEARCH".
My device wasn't discovered because of the code:

```
if(request.indexOf('M-SEARCH') > 0) { 
....
  respondToSearch();
}
```

Because 'M-SEARCH' is at index 0 (in my case) the above if statement is never true and respondToSearch() is never invoked. I've changed the code to:

```
if(request.indexOf('M-SEARCH') >= 0) { 
....
  respondToSearch();
}
```

and my Echo has been discovered successfully.

This could be the fix for those with devices that are not discovered.

tonywye, 2017-12-30:
I have just an Echo 2 and its now working for me when I make the changes above.Good stuff !
ToWy

hinkey247, 2017-12-30:
Hi tonywye - been spending days trying to get this to work as new to alexa 2nd gen - Please could you give full instructions to get this to work THANK YOU :)

tonywye, 2017-12-31:
@hinkey247 this .ino is working on my ESP8266-12, when I discover devices it finds 1 device and "Alexa turn on light" makes GPIO4 go on. There was no need to enable the Wemo skill on Alexa.
[Alexa-test_Echo_2_add_wifi_info.ino.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1594887/Alexa-test_Echo_2_add_wifi_info.ino.txt)

hinkey247, 2017-12-31:
Thanks :) Ive been pulling my hair out trying to get this to work.

hinkey247, 2017-12-31:
ps could you send this file as .ino thanks

hinkey247, 2017-12-31:
HI tonywye - just got this working - A BIG THANK YOU :)

hotelafrica, 2017-12-31:
I am still pulling my hair for this... I have echo 2nd gen. I've made sure that all the stuff commented above and also tried all the ino files above. Still nothing, Alexa can't discover the it.
Here's my current setup. Hopefully someone can give me a direction so I can sort this out... Q.Q Thanks in advance...

- ESP8266 NodeMCU LUA CP2102 ESP-12E (From Amazon, https://www.amazon.com/dp/B010O1G1ES/_encoding=UTF8?coliid=I2V7FJXT6E2L92&colid=2FBD4UXK743JT&psc=1)

- Linksys WRT AC1900AC router (i've done factory reset just in case)
- uPnP is enabled & multicast filter option is unchecked in the router setting.
- no 'wemo' skill is enabled

here's what I get from Serial monitor....:
192.168.1.55, port 58631
192.168.1.55, port 51710
192.168.1.55, port 51710
192.168.1.55, port 51710
192.168.1.55, port 51710
192.168.1.55, port 58631
192.168.1.57, port 52543
192.168.1.57, port 52543
192.168.1.57, port 52543
192.168.1.57, port 52543
192.168.1.55, port 58631
192.168.1.85, port 50000
192.168.1.85, port 50000
Responding to search request ...

Sending response to 192.168.1.85
Port : 50000
Response sent !
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>light</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to eventservice.xml ... ########

 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>light</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

192.168.1.85, port 50000
192.168.1.85, port 50000
Responding to search request ...

Sending response to 192.168.1.85
Port : 50000
Response sent !
########## Responding to  /upnp/control/basicevent1 ... ##########
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
192.168.1.85, port 50000
192.168.1.85, port 50000
Responding to search request ...

Sending response to 192.168.1.85
Port : 50000
Response sent !
########## Responding to  /upnp/control/basicevent1 ... ##########
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
192.168.1.55, port 6641
Responding to search request ...

Sending response to 192.168.1.55
Port : 6641
Response sent !
192.168.1.55, port 6641
Responding to search request ...

Sending response to 192.168.1.55
Port : 6641
Response sent !

hinkey247, 2017-12-31:
Hi mate I will email you later today and give you help I have now got my alexa 2nd gen to work thanks to this website.

   

Sending :urn:Belkin:device:controllee:1lightBelkin International Inc.Socket3.1415Belkin Plugin Socket 1.0
uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f221517K01017690urn:Belkin:service:basicevent:1urn:Belkin:serviceId:basicevent1/upnp/control/basicevent1/upnp/event/basicevent1/eventservice.xml

Sending :urn:Belkin:device:controllee:1lightBelkin International Inc.Socket3.1415Belkin Plugin Socket 1.0
uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f221517K01017690urn:Belkin:service:basicevent:1urn:Belkin:serviceId:basicevent1/upnp/control/basicevent1/upnp/event/basicevent1/eventservice.xml

request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1">1</u:GetBinaryState></s:Body></s:Envelope>

request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1">1</u:GetBinaryState></s:Body></s:Envelope>

jerrishjacob, 2017-12-31:
Thanks @tonywye @kakopappa for all the guidance. I have been able to get it working on my 2nd generation and nodemcu esp8266.

hkauroberoi, 2017-12-31:
None of the code is working in my case. I have an existing ESP installed under a switchboard since last two months and it is working fine. However, with the new ESP boards that I purchased, Alexa can't discover and there's no activity in the serial console.... Used previously tested code and copied from here also but to no avail... What could be the issue??

hinkey247, 2017-12-31:
Hi hkauroberoi - could the problem be the alexa has the new firmware update? but cant understand why no activity in serial console - surely it must not be flashing right.

hkauroberoi, 2017-12-31:
My bad. Im using the same alexa dot with which my previous esp is working. The code flashes perfectly and as im also using mqtt, the mqtt messages are being received normally.  It is just alexa detection which shows no activity in the console. I suspect firmware as till last week on the same esp the detection worked. Would run packet capture to see what's going on.

MacSass, 2017-12-31:
Hello,
let me try to get some structure to this, because I think there are some confusing things and we (including myself) have not been very exact in our communication so far.

There are two "types" of Echo devices out there currently obviously - but they different types are NOT aligned to the different Echo "generations". So instead of speaking of "generation xy" let´s talk about "type 1" or "type 2".

The "type 1 = old style" devices have been working in the past. Currently in the field they are the
Echo Dot generation 1
Echo Dot generation 2
Echo generation 1

The devices that came out end of this year are the "type 2" devices - namely
Echo generation 2
Echo Plus

For Echo Show and Echo Spod I do not know what "type" they are, but I suspect they will be "type 2" as thy are very new or just releasing now. I´m not sure if Amazon will update the "type 1" devices to the new type by firmware in the future, so far my Echo Dot still works the "old style".

What is the difference between "type 1" and "type 2"?
"Type 1" devices during discovery look for a response with ""urn:Belkin:device:**""
"Type 2" devices during device discovery look for a response with "upnp:rootdevice"

I´m currently looking into the fact if there is also a difference in what the devices send for a switch command - I think so.

I have a slightly modified code running here now that works with my "type 2 Echo Plus" for discovery and switching with no problem, but I think it does not work for my "type 1 Echo Dot Gen2" anymore. So we might have to make the code to react to both "types" which it does not right now if I´m not mistaken.

I´ll do some more testing as I have both "types" available and come back with final conclusion hoping that kakopappa will be able to fix it, so it works for all of us ...

Regards - MacSass

MacSass, 2017-12-31:
@hotelafrica 
If you have a "Echo Gen 2" (the newer, smaller version of the Echo, but not Echo Dot) then this is a "type 2" device looking for a response with "upnp:rootdevice", while your log shows that your ESP sends the "type 1" response.
It should work for you! if you change
"USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n" in **_switch.cpp_** to
"USN: uuid:" + persistent_uuid + "::upnp:rootdevice\r\n"
that´s what I did and what makes it working with my Echo Gen 2 ( = type 2 device).

As said - we might need code that "adapts" to the two types of devices out there right now to make it work for all at the same time.

Regards - MacSass

hinkey247, 2017-12-31:
Hi your right - I am a new comer to this system got a new alexa 2nd gen this Christmas and thought all the devices were the same ie languages / setup. I was under false information to watch youtube videos and hey presto it works, but it didn't. and trying to find answers why for 3/4 days was a nightmare only to stumble on this this forum to get the answers. after downloading the new .ino file it worked perfectly - this was on the understanding there was a difference between the dot and alexa 2nd gen plus firmware update.

sydknee, 2017-12-31:
I'm having real problems with this and being new to all this too isn't helping.

I've got the Echo Plus (two dots on their way when they are back in stock)

I first tried someone else's adaptation of this to control 4 relays and couldn't find the devices, tried the source code from Kakopappa and still no joy, found this forum and made some changes to the source code and got it working (well couldn't fully test it as hadn't put anything on the connections).  Deleted the items from Alexa and made the changes to the 4 relay version and nothing.  Gave up at that point as had to go to work.

Luckily I had saved the changed versions.

Came back this morning and loaded up the amended source code and nothing.

Tried everything today, amending all the different versions of the code and I get various different outputs on the serial monitor.  Sometimes nothing and sometimes loads of code.

It would be interesting to see if the real Belkin devices have the same issue but they are a bit overpriced to try.

hotelafrica, 2017-12-31:
@MacSass 
Yes, i've tried that change previously and didn't work. Well, I did it again few minutes ago and here's the serial monitor result. it does get "::upnp:rootdevice\r\n" from Echo and response code is kicking off.
I wasn't sure if it was the router that's blocking the multicast and tried to use different router, still no luck.

here's the serial monitor output...

⸮⸮-s⸮⸮
Connecting to WiFi
Connecting ........
Connected to JJJJ_2.4GHz
IP address: 192.168.1.89
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 
80
WebServer started on port: 
81
Adding switches upnp broadcast responder
Adding switch : computer index : 0
Adding switch : kitchen lights index : 1
Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>computer</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f-80</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>kitchen lights</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f-81</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to eventservice.xml ... ########

 ########## Responding to eventservice.xml ... ########

Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
[... 20 lines cut ...]
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>1</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

########## Responding to  /upnp/control/basicevent1 ... ##########
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>0</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

########## Responding to  /upnp/control/basicevent1 ... ##########
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>1</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

charliebrowndog, 2017-12-31:
My Echo Gen 2 (late 2017) would not recognize my Mini D1 until I changed "urn:Belkin:device:**" to ---> "upnp:rootdevice" in the "fauxmoESP.h" file (line 37).  Recompiled and it was recognized!  Hope this helps someone else...

jerrishjacob, 2018-01-01:
@tonywye , I used your link "Alexa-test_Echo_2_add_wifi_info.ino" and I have got the 2nd generation echo working with esp8266 nodemcu. Need some more help. 
1) How do I control more than one relay on a single esp8266 nodemcu. 
2) How do I introduce more than one esp8266 into the network.
Thanks

hinkey247, 2018-01-01:
Hi just trying to sort this code out for myself to run 8 relays / going to use youtube 8 relay code with this one a bit of of a miss mash but if you bet me upload yours if not I will / going to mess around later today. Good luck. Martin

hinkey247, 2018-01-01:
Ps at least we have the two systems talking to each other / that's a start. 😆

 
  On Mon, 1 Jan 2018 at 1:12, MARTIN HINKS<martinhinks@btinternet.com> wrote:   Hi just trying to sort this code out for myself to run 8 relays / going to use youtube 8 relay code with this one a bit of of a miss mash but if you bet me upload yours if not I will / going to mess around later today. Good luck. Martin

hkauroberoi, 2018-01-01:
If this helps someone.... The alexa dot firmware is : 597464620

**I got it working**.... First tried with making changes to switch.cpp (belkin to root) but in vain. Virtually tried everything/every code but there was no activity in the serial console. Then checked the settings in router and found out that UPNP was unchecked. Enable the UPNP in router settings and used previous code (fauxmo) - all 7 devices successfully detected.

`*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.100
{"device_id":"2C:3A:E8:35:D0:E7","device_name":"Device1"}Attempting MQTT connection...connected
HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Mon, 22 Jun 2015 17:24:01 GMT
EXT:
LOCATION: http://192.168.0.100:52003/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 44455635D0E703
SERVER: Unspecified, UPnP/1.0, Unspecified
X-User-Agent: redsonic
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-44455635D0E703::urn:Belkin:device:**

HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Mon, 22 Jun 2015 17:24:01 GMT
EXT:
LOCATION: http://192.168.0.100:52003/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 44455635D0E703
SERVER: Unspecified, UPnP/1.0, Unspecified
X-User-Agent: redsonic
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-44455635D0E703::urn:Belkin:device:**

HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Mon, 22 Jun 2015 17:24:01 GMT
EXT:
LOCATION: http://192.168.0.100:52004/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 44455635D0E704
SERVER: Unspecified, UPnP/1.0, Unspecified
X-User-Agent: redsonic
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-44455635D0E704::urn:Belkin:device:**

HTTP/1.1 200 OK
CACHE-CONTROL: max-age=86400
DATE: Mon, 22 Jun 2015 17:24:01 GMT
EXT:
LOCATION: http://192.168.0.100:52005/setup.xml
OPT: "http://schemas.upnp.org/upnp/1/0/"; ns=01
01-NLS: 44455635D0E705
SERVER: Unspecified, UPnP/1.0, Unspecified
X-User-Agent: redsonic
ST: urn:Belkin:device:**
USN: uuid:Socket-1_0-44455635D0E705::urn:Belkin:device:**

`

hinkey247, 2018-01-01:
Great news. 😀

MacSass, 2018-01-01:
Happy new year everybody, new year - new testing :-)
and I can confirm that I have working code for all types of devices!

I have done some extensive testing today with 3 different Echo devices:
Echo Dot Gen 2 (FW 597464620) - Type 1  
Echo Generation 2 (FW 595530420)- Type 2
Echo Plus - Type 2
**_Please read my post above on "type 1" vs. "type 2" devices._**

Please also take not that I´m only referring to the code listed here in the MASTER branch from "kakopappa" - not any other code that was referenced before (e.g. those that have files like "fauxwemos", etc.). 
I´m referring to https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch only!

The code as it is right now is working fine for **type 1** devices with no modification. Discovery of devices as well as switching works fine!
For **_type 2_** the code currently does NOT work, it does not show any devices after discovery. Symptom is that in serial monitor you only get:

> Sending response to 192.168.1.28
> Port : 50000
> Response sent !

while for **type 1** devices you additonally see this during discovery:

> Sending response to 192.168.1.28
> Port : 50000
> Response sent !
> ########## Responding to setup.xml ... ########
> .....

The reason is that **type 2** devices look for a response of **_urn:Belkin:device:**_** and if that does not happen they do not query for the services ... thus ESP never responds.

I can make the code work with **type 1 and type 2** devices by making the following change that I already mentioned in a previous post. Change one line close to the bottom of **_switch.cpp_** from:
"USN: uuid:" + persistent_uuid + "::urn:Belkin:device:**\r\n" 
to be:
"USN: uuid:" + persistent_uuid + "::upnp:rootdevice\r\n"

After this your code should work with all types of Echo devices and discovery as well as switching should work find. This is working in my home right now. I have ran several discoveries with only type 1 or only type 2 devices on the network with consistent results.

As I am not experienced in GIT usage I´m not sure how to propose that change for testing to Kakopappa, thus I will contact him directly to see what he thinks. Maybe he can open a temporary branch that you guys can test to confirm my finding.

Regards - MacSass

PS: As Amazon Echo does use UPNP discovery with UDP broadcasts obviously you need to have UPNP enabled in your router and not eventually blocked by it!

hotelafrica, 2018-01-01:
it looks like i am hitting the dead end...
I did everything above and no luck. Alexa just can't find the esp8266.
Not sure if this is because i'm using ESP8266 NodeMCU, not wemo-mini. Or because of some kind of network setting or hardware. (TP-link smart outlet works fine through Echo)
[image]

I am waiting for a raspberry pi to be delivered and see if I can utilize my esp8266 via mqtt...

here's my serial output, (Echo is 192.168.1.85)

Connecting to WiFi
Connecting .......
Connected to JJJJ_2.4GHz
IP address: 192.168.1.89
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 
80
WebServer started on port: 
81
Adding switches upnp broadcast responder
Adding switch : office lights index : 0
Adding switch : kitchen lights index : 1
Got UDP Belkin Request..

Sending response to 192.168.1.55
Port : 61196
Response sent !

Sending response to 192.168.1.55
Port : 61196
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.1.55
Port : 61196
Response sent !

Sending response to 192.168.1.55
Port : 61196
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
Got UDP Belkin Request..

Sending response to 192.168.1.85
Port : 50000
Response sent !

Sending response to 192.168.1.85
Port : 50000
Response sent !
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>kitchen lights</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f-81</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>office lights</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f-80</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

 ########## Responding to eventservice.xml ... ########

[... 72 lines cut ...]
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

########## Responding to  /upnp/control/basicevent1 ... ##########
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>1</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

Got UDP Belkin Request..

Sending response to 192.168.1.55
Port : 52823
Response sent !

Sending response to 192.168.1.55
Port : 52823
Response sent !
 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>office lights</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e651b70f-80</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

jcarley, 2018-01-01:
@hotelafrica from looking at your code, I think your response message is wrong.  I got this working with my Echo Plus (firmware 595530520)  Here's what I have in Switch.cpp, method Switch::respondToSearch.  Pay close attention to the third to last line and the second to last line.  Both contain the string "upnp:rootdevice".  That is literally all the changes I made.  I upload my sketch to a Node MCU, and it worked.  Echo found the devices no problem.

`String response = 
       "HTTP/1.1 200 OK\r\n"
       "CACHE-CONTROL: max-age=86400\r\n"
       "DATE: Sat, 26 Nov 2016 04:56:29 GMT\r\n"
       "EXT:\r\n"
       "LOCATION: http://" + String(s) + ":" + String(localPort) + "/setup.xml\r\n"
       "OPT: \"http://schemas.upnp.org/upnp/1/0/\"; ns=01\r\n"
       "01-NLS: b9200ebb-736d-4b93-bf03-835149d13983\r\n"
       "SERVER: Unspecified, UPnP/1.0, Unspecified\r\n"
       "ST: upnp:rootdevice:**\r\n"
       "USN: uuid:" + persistent_uuid + "::upnp:rootdevice:**\r\n"
       "X-User-Agent: redsonic\r\n\r\n";`

MacSass, 2018-01-01:
@hotelafrica:
This is strange - find attached my serial output. 
I notice that in your output the XML tags are missing, but I guess those have been lost posting to github?
Hope my attached log helps - it was done with an ESP-12 development board (it really should work regardless of what ESP8266 board you use, I use several different ones).

Also, instead of pasting long logs into this thread **_better attach a .txt file with it_** - it keeps the thread less cluttered and makes sure nothing gets stripped. If you can attach our log as a file I can compare the files to see if there is any difference. 

Regards - MacSass
[ESP serial log.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1596524/ESP.serial.log.txt)

hotelafrica, 2018-01-02:
@jcarley 
I applied your change and tried again. No detection happening still.

@MacSass 
the symbol you are refering as '#' is actually different form of backslash
(it actually looks like W with strike through)
It's because I've installed Korean input language in my windows 10 and language setting for 'Non-unicode programs' is also set as Korean.

But since you mentioned about it, i am going to remove all those settings and maybe reinstall the arduino ide THEN try the code again. (Maybe Echo doesn't understand response because it's encoded differently?)

if it works that way, then i'll have to figure out different way to make it work with Korean set up in my computer (It's kind of must have for my daily use).

Let me play with it little more, I'll let you know!

hotelafrica, 2018-01-02:
[image]

Removed Korean from windows (but didn't reinstall the arduino yet) and tried again.

well... not working... :( 

let me reinstall the program and see what happens...

here's a log file after flashing the esp.

[espLog_0101.txt](https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/files/1596576/espLog_0101.txt)

Again, thank you guys for look at my problem, hopefully I can sort this out somehow!

cheers.

jcarley, 2018-01-02:
@hotelafrica Shot in the dark here.  See attached screen shot.

[image]

hotelafrica, 2018-01-02:
@jcarley
I did, no discovery.
My next step is to try again with different router.
I kind of already tried it by using a hotspot from my iphone but let me do it again with actual router.

[59 more replies]