kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #23: unable to find devices

gkrananth, 2017-11-10:
.
Connected to Ananth
IP address: 192.168.1.25
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port: 
80
WebServer started on port: 
81
Adding switches upnp broadcast responder
Adding switch : office lights index : 0
Adding switch : kitchen lights index : 1
 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: urn:schemas-upnp-org:device:InternetGatewayDevice:1
USN: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E::urn:schemas-upnp-org:device:InternetGatewayDevice:1

 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E
USN: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E

B4299E::urn:schemas-upnp-org:device:InternetGatewayDevice:1

 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: upnp:rootdevice
USN: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E::upnp:rootdevice

4299E

B4299E::urn:schemas-upnp-org:device:InternetGatewayDevice:1

 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: urn:schemas-upnp-org:service:Layer3Forwarding:1
USN: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E::urn:schemas-upnp-org:service:Layer3Forwarding:1

ce:1

 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: urn:schemas-upnp-org:device:WANDevice:1
USN: uuid:C6D05FCA-D445-804C-AC3B-2337D607AF50::urn:schemas-upnp-org:device:WANDevice:1

Forwarding:1

ce:1

 request  : M-SEARCH * HTTP/1.1
[... 390 lines cut ...]
 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: urn:schemas-upnp-org:service:Layer3Forwarding:1
USN: uuid:D231DD98-C695-B6C7-BFDA-3F84B5B4299E::urn:schemas-upnp-org:service:Layer3Forwarding:1

ce:1

 request  : NOTIFY * HTTP/1.1
Host: 239.255.255.250:1900
Cache-Control: max-age=60
Location: http://192.168.1.1:1780/InternetGatewayDevice.xml
NTS: ssdp:alive
Server: POSIX, UPnP/1.0 linux/5.100.138.11
NT: urn:schemas-upnp-org:device:WANDevice:1
USN: uuid:C6D05FCA-D445-804C-AC3B-2337D607AF50::urn:schemas-upnp-org:device:WANDevice:1

Forwarding:1

ce:1

Please find above the request there is no urn for Belkin recevied.  Please let me know what was the issue.  I am trying with Echo Gen2

canaokar, 2017-11-13:
I am having the same issue

kanhaic, 2017-11-20:
+1 same issue here

kakopappa, 2017-11-21:
There is a problem with Echo Gen2 with the latest firmware (591448720) according to @MacSass
https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/22

I am working on a smart home skill and it is not ready yet. I will update this issue when it's ready.