kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #25: Does hot handle GetBinaryState message

riban-bw, 2017-11-19:
The Alexa app allows control of devices. It sends a GetBinaryState message to find the current status. This is erroneously interpreted as a switch ON command. Please add handler for GetBinaryState messages. Format of request is same as SetBinaryState message.

kakopappa, 2017-11-27:
https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066

can you give this a try and let me know.

riban-bw, 2017-11-27:
That _almost_ works. You need to add a condition around the SetBinaryState, e.g.
```
if(request.indexOf("SetBinaryState") >= 0) {
        if(request.indexOf("<BinaryState>1</BinaryState>") >= 0) {
            Serial.println("Got Turn on request");
            turnOnRelay();
        }
  
        if(request.indexOf("<BinaryState>0</BinaryState>") >= 0) {
            Serial.println("Got Turn off request");
            turnOffRelay();
        }
      }
```
Also, although the code works, sendRelayState() should use u:GetBinaryStateResponse tag, not u:SetBinaryStateResponse tag.

You don't need the trailing "\r\n" in the GetBinaryStateResponse.

I am not sure why my attempts to implement this failed!

kakopappa, 2017-11-27:
Can you send me the working full code pls?

riban-bw, 2017-12-02:
Sorry for delay - two of my ESP8266 modules died! The code works but I had to make some changes for my device:

`
const int relayPin = D1; //D1 is not recognised as a constant by most ESP8266 boards in Arduino. Suggest using universal constant
`

connectWifi() does not connect to wifi. I used WiFiManager to get it working. I don't know why your code does not work for me. (Yes I did set SSID / password!)

tommyfranke, 2017-12-03:
Hi all. 
@kakopappa, thank's for the great work.
The above code works wonderful on my new Amazon Dot V2 with actual Firmware 595459620.
My Nodemcu was found at once, you don't need the Wemo-Skin!!!

@riban-bw 
you have to increase the wait time of 5 seconds in connectWifi() to about 10 or more  seconds. 
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    if (i > 20){          //--------------------increase to your needs
      state = false;
      break;
    }
    i++;
  }

kakopappa, 2017-12-04:
thanks,  code updated.