kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #28: setup.xml not declared error

ianc99, 2017-12-07:
Hi,
I'm getting "Switch.cpp:214: error: 'setup_xml' was not declared in this scope" error when trying to compile.
So - I just want to check if I've pulled down the right version which has the ability to respond to Alexa status requests.  Or am I doing something stupid :-)

Thanks

Antoinebr, 2017-12-10:
I also have the same issue  

```
sketch/Switch.cpp: In member function 'void Switch::sendRelayState()':
Switch.cpp:214: error: 'setup_xml' was not declared in this scope
   Serial.println(setup_xml);
                  ^
exit status 1
'setup_xml' was not declared in this scope

```

ianc99, 2017-12-10:
I just replaced it with
` Serial.println(body);`
and it worked

kakopappa, 2017-12-19:
sorry about that guys. I have updated the code now.

closed 2017-12-19 by kakopappa