kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #33: Dimmer Module

Braandn, 2017-12-28:
Is it possible to add a Dimmer Module to the system, so that you can say "Alexa set the light to 20%"
if yes, could you post an example?

kakopappa, 2017-12-30:
Sorry, dimmer modules are not supported. This project supports only turning on/off. 

To Alexa set the light to xxxx you have to make an Amazon Skill. 

I am in the process of building a smart-home-skill for everyone and about 90% of the work is done now. I will update this issue when it's completed.

kakopappa, 2018-01-05:
I just received the confimation from Amazon that my skill is live. 

Can  you check for smart home skill called "sinric" ?

register at sinric.com

1. create a new device 
2. select light, name "bedroom lights", 
3. alexa dim "bedroom lights"

kakopappa, 2018-01-05:
Hello 
Past few weeks I have been working on a smart home skill and a website  which allows you add development boards such as WEMOS D1 to Alexa eco system without emulating as a belkin switch.  Mainly, because some people have issues in the UDP broadcast and different Alexa devices such as ECO, DOT has different XML formats and this is logic is quite difficult to code. So, I belive moving forward using the smart home skill is the right thing to do;

So, I would like to invite everyone to give it a try. 

1. Goto sinric.com

2. Create an account

3. Use the API key displayed on the home screen to connect your WEMOS D1 to sinric. Here is the code . Just replace the API KEY
https://github.com/kakopappa/sinric/blob/master/arduino_sinric_example.ino

4. Install smart home skill from 
https://www.amazon.com/dp/B078RGYWQQ/ref=syps?s=digital-skills&ie=UTF8&qid=1515150398&sr=1-18&refinements=p_n_date%3A14284925011

4.1 Link, eneter you login details to sinric.com

5. Discover. 

6. Rate the Skill so other people can easily find it!

You can turn on/off same like emulating a belkin device. 

No need to say like Alexa ask sinric to turn on bedroom lights. Simple as a Alexa turn on bedroom lights.

If you have any issue or comments report it here;
https://github.com/kakopappa/sinric

Thanks
kakopappa

Aircoookie, 2018-01-09:
@Braandn I've developed a sketch called Espalexa (https://github.com/Aircoookie/Espalexa) based on kakopappa's single switch (https://github.com/kakopappa/arduino-esp8266-alexa-wemo-switch) project. It might suit your needs since it supports dimming, however only with one switch (could quite easily be extended for more, though).

This works because it uses the Philips Hue protocol instead of the Belkin Wemo one.
Won't create a PR though since it is too different from this project.

Thank you, @kakopappa ! wouldn't have been able to get this to work without your code as a foundation.

MacSass, 2018-01-21:
Hi,
while kakopappa is working on the Alexa Skill, which might provide the smoothes road for the future - I can really recommend the current implementation from Aircookie - he has added support for dimming as well as multiple devices (emulates hue) - works perfectly for me.
Regards - MacSass

arman4455, 2018-03-19:
@kakopappa Thanks! I got sinric to work with the example blind code. would you be able to make some 
 code based on that so i can open and close the blinds? essentially control the servo?

Thanks, 
Arman

rramesh68, 2020-03-08:
@kakopappa I tried downloading the switch_example.ino and when I try to compile it i get the following error "json' was not declared in this scope" against this line  String deviceId = json ["deviceId"];.

I am a novice. Kindly advise on the way forward.

Thanks
Ramesh

kakopappa, 2020-03-09:
You are missing the json library. Install ArduinoJson library first