kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #35: ESP (NodeMCU) crashes when adding the 4th switch...

BlackyS, 2018-01-04:
Hello!
First of all: thank you very much for sharing your code with us!!!
Unfortunately I'm facing a serious issue when adding my 4th swith to the code.
3x Switches are running very smooth, but after adding the 4th my ESP crashes with:

Exception (28):
epc1=0x40203bfb epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

Is it possible that there is an issue with heap consumption?
I'm starting with: 
Adding switch : eins index : 0  => loop heap size: 21248
Adding switch : zwei index : 1  => loop heap size: 21296
Adding switch : drei index : 2   =>loop heap size: 21216

and once I enter the loop (): =>  ... loop heap size: 6584

Any idea what could cause that crash after adding the 4th switch?
BR
Martin

kakopappa, 2018-01-05:
Past few weeks I have been working on a smart home skill and a website  which allows you add development boards such as WEMOS D1 to Alexa eco system without emulating as a belkin switch.  Mainly, because some people have issues in the UDP broadcast and different Alexa devices such as ECO, DOT has different XML formats and this is logic is quite difficult to code. So, I belive moving forward using the smart home skill is the right thing to do;

So, I would like to invite everyone to give it a try. 

1. Goto sinric.com

2. Create an account

3. Use the API key displayed on the home screen to connect your WEMOS D1 to sinric. Here is the code . Just replace the API KEY
https://github.com/kakopappa/sinric/blob/master/arduino_sinric_example.ino

4. Install smart home skill from 
https://www.amazon.com/dp/B078RGYWQQ/ref=syps?s=digital-skills&ie=UTF8&qid=1515150398&sr=1-18&refinements=p_n_date%3A14284925011

4.1 Link, eneter you login details to sinric.com

5. Discover. 

6. Rate the Skill so other people can easily find it!

You can turn on/off same like emulating a belkin device. 

No need to say like Alexa ask sinric to turn on bedroom lights. Simple as a Alexa turn on bedroom lights.

If you have any issue or comments report it here;
https://github.com/kakopappa/sinric

Thanks

slavino, 2018-02-22:
For me work 4 switches just fine on NodeMCU 0.9. Just to make sure - did you change the listen port and introduced the methods at top of *.ino ?

`
lvngRmLEDStripe = new Switch("LED stripe in living room", 80, lvngRmLEDStripeOn, lvngRmLEDStripeOff);

lvngRmCornerLamp = new Switch("corner lamp in living room", 81, lvngRmCornerLampOn, lvngRmCornerLampOff);

lvngRmSubwoofer = new Switch("subwoofer in living room", 82, lvngRmSubwooferOn, lvngRmSubwooferOff);

lvngRmSocket4 = new Switch("socket 4 in living room", 83, lvngRmSocket4On, lvngRmSocket4Off);
`