kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #36: Unintended (emulated WeMo switch) activity if going to the Alexa Web App - Smart Home - Devices page

wmweis62, 2018-01-10:
Kakopappa, I am using your WeMo switch emulated code as the base code for a solution I developed for people with limited mobility. I provide them with the ability to voice control their hospital beds, because for them there is no other alternative. Everything necessary to make this work is my time and money as part of a volunteer effort.   

I noticed some undesirable behavior and I am wondering if this is something you also noticed and if you thought of a solution for it. (Nassir may have already contacted you regarding this, so sorry if this is duplicate)

In my solution, I have 4 WeMo emulated switches that control 4 movements of the bed. In my lab, when I discover devices I see those devices in the Devices webpage within the Amazon\Alexa web app. The issue is anytime I either open the Devices web page, or refresh the Devices web page, or click Discover in the Devices web page all 4 relays cycle on and off. The downside of this is unintended bed motion creating a possible safety issue.

Looking at the Serial Port log, I think I see the cause but I have not figured out what to do about it.  Associated with the Unintended switch cycling is a GetBinaryState message that will happen when you simply open the Devices web page, or refresh the Devices web page, or click Discover on the Devices webpage <even when the devices are already known and online>.  Associated with the desired switch activity is a SetBinaryState message. I contacted Amazon but don't see them doing anything  about it. I also contacted WeMo just in case their engineering team had to deal with this already since testing with a physical WeMo switch does not exhibit this same behavior.

Here are some snippets from the serial port for a few scenarios - 

This is what a Voice Command for Turn on relay 2 looks like: (This is desirable behavior)
 
########## Responding to  /upnp/control/basicevent1 ... ##########
"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:SetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:SetBinaryState></s:Body></s:Envelope>
 
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:SetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:SetBinaryState></s:Body></s:Envelope>
Got Turn on request
Switch 2 turn on ...
 
 
**And here is what it looks like when you click discover new devices for the same switch even when the devices are known and online (Undesirable)**
 
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got Turn on request
Switch 2 turn on ...########## Responding to  /upnp/control/basicevent1 ... ##########
"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
 
And this is what it looks like if you simply refresh the Devices page in the Amazon\Alexa webpage under Smart Home  (Undesirable)
 
request:"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got Turn on request
Switch 2 turn on ...########## Responding to  /upnp/control/basicevent1 ... ##########
"1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>

wmweis62, 2018-01-10:
Happy to report testing is going well for the fix for handling GetBinaryState messages.

kakopappa, 2018-01-11:
Did you get it fixed? Let me know if it needs any code changes.

Also try sinric smart home skill. It should be available in us

wmweis62, 2018-01-11:
Thanks for checking, yes all testing has gone well so far.

wmweis62, 2018-03-18:
I apologize for posting this here, but for now I am at a loss as to the best place for this. I am using the core sketch developed by kakopappa and others which allows for some callbacks to control some wemo emulated devices. I use this code in a very impactful way and would like to thank the developers. I have created voice activated controllers for beds and now doors so people with limited mobility can gain greater independence and I give those solutions to them at not cost. (http://limitedmobility.solutions)

The problem I am seeing is related to this core sketch although I am not suspecting a problem with the sketch. I raise this point here because I am at a loss with how to troubleshoot this problem and perhaps those more familiar with the way the messages from Amazon work in conjunction with this sketch can shed some light.

So finally, here is the problem. The sketch is loaded on an ESP8266 NodeMCU E12 and it has 2 emulated WeMo switches. I connect my laptop to the ESP8266 and watch the serial port. When the upload finishes I observe the serial monitor and can see it obtain a IP address and see the following:
Connecting to WiFi
Connecting ....
Connected to CenturyLink2749
IP address: 192.168.0.101
Begin multicast ..
Udp multicast server started at 239.255.255.250:1900
WebServer started on port:
80
WebServer started on port:
81
Adding switches upnp broadcast responder
Adding switch : @@@@ ::getAlexaInvokeName @@@
FrontDoor index : 0
Adding switch : @@@@ ::getAlexaInvokeName @@@
BackDoor index : 1

The next step is to issue a Discovery using either a voice command or a discovery initiated by hitting the button in the Alexa app. During Discovery we should see additional text on the serial monitor that looks like the following. (At this lady’s house, discovery of this device never works and no additional text appears on the serial monitor during discovery. This exact same sketch works EVERYWHERE else with no problem.) So my question is this, how do I track where the UDP and the other are getting lost and I have no clue how to trace it. Again during Discovery I should see the UDP and other messages on the serial monitor, but at this DSL connected house I see none of these.

Got UDP Belkin Request..

Sending response to 192.168.0.113
Port : 50000
Response sent !

Sending response to 192.168.0.113
Port : 50000
Response sent !
########## Responding to setup.xml ... ########

localiP:
192.168.0.101
Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>FrontDoor</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Emulated Socket</modelName><modelNumber>3.1415</modelNumber><UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e65e33ac-80</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

########## Responding to setup.xml ... ########

localiP:
192.168.0.101
Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>BackDoor</friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Emulated Socket</modelName><modelNumber>3.1415</modelNumber><UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e65e33ac-81</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

kakopappa, 2018-03-18:
I have seen issues like this often lately. It's something to do with router
blocking the UDP or new firmware changes

This is why I made Sinric smart home skill. I highly recommend you to move
to Sinric smart home skill

Checkout https://sinric.com

wmweis62, 2018-03-29:
Unless I am seeing a glitch in Amazon’s service today, it appears that as of today Amazon is requiring Skills for device discovery now. Devices I have been able to discover without a Skill enabled for the past 6 months are no longer being discovered, even though the serial monitor is showing the UDP message and all the normal messages that follow.  If this is true, we are going to have to rewrite some solutions we developed for people with limited mobility that we gave to them at no cost.

wmweis62, 2018-03-30:
It seems that discovery is working again and devices in sketches without Sinric are being discovered as they always have been. As I mentioned, discovery has always worked in my home lab, until yesterday afternoon. I saw all the normal activity on the serial monitor related to discovery including the UDP message and the subsequent messages, so I made the assumption Amazon was going to start enforcing Skills perhaps as added security so someone could not simply put a rouge device on someone’s network and have it get discovered.

wmweis62, 2018-03-30:
Discover devices is behaving ‘normal’ again Not sure what happened yesterday afternoon. Perhaps Amazon was testing a software change intended to make it tougher for rouge devices to be placed on people’s networks. (Rough devices could not get discovered unless a skill was also enabled, tougher for the bad guys to pull off)

From: Bill Weis
Sent: Thursday, March 29, 2018 5:13 PM
To: 'kakopappa/arduino-esp8266-alexa-multiple-wemo-switch' <reply@reply.github.com>
Subject: RE: [kakopappa/arduino-esp8266-alexa-multiple-wemo-switch] Unintended (emulated WeMo switch) activity if going to the Alexa Web App - Smart Home - Devices page (#36)

Unless I am seeing a glitch in Amazon’s service today, it appears that as of today Amazon is requiring Skills for device discovery now. Devices I have been able to discover without a Skill enabled for the past 6 months are no longer being discovered, even though the serial monitor is showing the UDP message and all the normal messages that follow.  If this is true, we are going to have to rewrite some solutions we developed for people with limited mobility that we gave to them at no cost.

uDude, 2018-05-22:
Bill Weis ... did you work at LA?  If so, PM  me.

wmweis62, 2018-05-22:
No I did not, sorry.

From: uDude
Sent: Tuesday, May 22, 3:39 AM
Subject: Re: [kakopappa/arduino-esp8266-alexa-multiple-wemo-switch] Unintended (emulated WeMo switch) activity if going to the Alexa Web App - Smart Home - Devices page (#36)
To: kakopappa/arduino-esp8266-alexa-multiple-wemo-switch
Cc: wmweis62, Author

Bill Weis ... did you work at LA? If so, PM me.
-