kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #37: Device appears to be malfuctioning.

emnik, 2018-01-25:
Hi! I have an amazon echo gen 2 and I was using this code with an esp for my light. Everything was working quite good but after each succesfull command Alexa was late responding Ok (or "I'm not sure what went wrong" although nothing went wrong!!!). 

Yesterday the firmware of Alexa was updated to 599472520. The commands are still successful issued and the relay opens and closes as it is supposed but most of the times the always late response is "this device appears to be malfunctioning".

So there is something wrong with the new firmware update and the status report that we send to Alexa.

Similar problem after the update I have with the espalexa code where when I set a light to off Alexa responds that this value is not supported by the device!!! Just before the update it worked perfectly...

flamboo, 2018-02-03:
It looks like it is not updating the Binary State when it sends the status to alexa.

kakopappa, 2018-02-06:
Not sure. try sinric.com

guedini, 2020-01-14:
I'm facing the same issue, did someone solve it?