kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #40: Works for a few minutes, then Alexa loses connection to device

Mehrtens, 2018-02-01:
I have successfully integrated this method into some of my existing Wemos code, and it works well. I have two devices connected to one Wemos, it boots up and connects to Wifi, I also have MQTT talking to a Pi, then the Alexa connection is successful. Alexa will function the devices on command, no problem. Then I leave and come back some time later, and find that while the Wemos is still connected to Wifi, Alexa no longer sees the devices. 
I am about to look into just getting Alexa to connect to my Pi as one device and just relay commands from that server. Any ideas? I am implementing the code just like the example. Thanks in advance!

kakopappa, 2018-02-01:
are you using the sinric skill or arduino-esp8266-alexa-multiple-wemo-switch code ? I have seen few people having this issues after updating Alexa firmware to the latest version

Mehrtens, 2018-02-01:
Not sure ... I am using the UpnpBroadcastResponder object model, I have an Echo Dot Gen 2. I noticed in another issue thread a link to a different implementation that looks like it does not use the UpnpBroadcastResponder , I was going to try this out and see...

Mehrtens, 2018-02-01:
https://gist.github.com/kakopappa/02299a8ceec0aaaad3ea3bd7f2af8066
This method? I will work on this tomorrow and see how it turns out.

kakopappa, 2018-02-01:
try sinric.com

Mehrtens, 2018-02-02:
Hi, found the example code and got the libraries, but not getting anything
with the site. Do you know if it is down? I see that I would need to sign
up and get an API key there right?

Thanks for the help

kakopappa, 2018-02-02:
@Mehrtens Site is up. 

it may take few seconds depending on the connection speed. There is a one angular file (bundle.js) which is little bit big.

Mehrtens, 2018-02-03:
My bad ... was using IE, switched to firefox and got right in. Got the code
working. Setup two Wemos devices, each with two relay outputs to control,
they work great. And no more dropouts, I come back hours later, and works
right away. This is the way to go for me.

In reviewing the sinric code, would I be correct in thinking that if I add
in the WebSocketEvent function additional code for handling if (action ==
getPowerState) {} that I could return back the status of the device relay
on request from Alexa, or is this a feature that is not available in the
Alexa Skill at this time? I don't suppose that source code is available, I
would like to attempt to add it...

Thanks for the help

kakopappa, 2018-02-03:
@Mehrtens I have updated the example code with more details now.
https://github.com/kakopappa/sinric/blob/master/arduino_examples/switch_example.ino