kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #46: Not an issue - a question

Ynot1, 2018-04-18:
New to the esp2866, and to alexa , but keen to learn a bit about both by playing with an integration between the two.
My esp2866's are en route from china, and my brand new echo plus is awaiting something real to interface to.
Is anyone succefully using this code or similar to interface to alexa using just the Wemo skill? I won't be able to use sinric as it not a released skill in NZ. Also not keen on building the Home automation stack at present. 
Thanks in advance.

kakopappa, 2018-04-18:
I am in the middle of diagnosing an issue related to Japanese with guys from Amazon. Once it's done I will release a version for AUS an NZ soon.

Ynot1, 2018-04-18:
Awesum! I shall watch this forum with interest for an update on the subject.
I didn't realise you created the sinric skill as well, it all kind of makes more sense now...