kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #48: How to make the state returned to Alexa from another gpio pin.

Ynot1, 2018-04-25:


kakopappa, 2018-04-25:
Don't think it's possible

I am not sure what are you trying to do. Check whether sinric.com can
support

Ynot1, 2018-04-25:
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <WiFiUdp.h>
#include <functional>
#include "switch.h"
#include "UpnpBroadcastResponder.h"
#include "CallbackFunction.h"

// prototypes
boolean connectWifi();

//on/off callbacks 
bool garageDoorOpenOn();
bool garageDoorOpenOff();
bool garageDoorCloseOn();
bool garageDoorCloseOff();

// Change this before you flash
const char* ssid = "JustBePatient_VirusDownloading";
const char* password = "HH563dggrtvvxz";

boolean wifiConnected = false;

UpnpBroadcastResponder upnpBroadcastResponder;

Switch *garagedooropen = NULL;
Switch *garagedoorclose = NULL;

bool isgarageDoorOpenOn = false;
bool isgarageDoorCloseOn = false;
bool GarageDoorState = false;
bool PrevGarageDoorState = false;

const int garageDoorRelay = 0;  // GPIO0 pin.
const int GPIO2 = 2;  // GPIO2 pin. this would be better if it were the RXD pin, it would make restarts with the door open possible...

void setup()
{
  pinMode(garageDoorRelay, OUTPUT);
  pinMode(GPIO2, OUTPUT); // to start with, it changes later...
     
  Serial.begin(9600);

  Serial.println("Booting...");
  delay(2000);

  //flash fast a few times to indicate CPU is booting
  digitalWrite(GPIO2, LOW); 
  delay(100);    
  digitalWrite(GPIO2, HIGH);
  delay(100); 
  digitalWrite(GPIO2, LOW); 
  delay(100);    
  digitalWrite(GPIO2, HIGH);
  delay(100);   
  digitalWrite(GPIO2, LOW); 
  delay(100);   
  digitalWrite(GPIO2, HIGH);
  
  Serial.println("Delaying a bit...");
  delay(2000);   
  
  // Initialise wifi connection
  wifiConnected = connectWifi();
     
  if(wifiConnected){

  //flash slow a few times to indicate wifi connected OK
  digitalWrite(GPIO2, LOW); 
  delay(1000);    
[... 127 lines cut ...]
  Serial.print("Connecting ...");
  while (WiFi.status() != WL_CONNECTED) {
    delay(5000);
    Serial.print(".");
    if (i > 10){
      state = false;
      break;
    }
    i++;
  }
  
  if (state){
    Serial.println("");
    Serial.print("Connected to ");
    Serial.println(ssid);
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  }
  else {
    Serial.println("");
    Serial.println("Connection failed. Bugger");
  }
  
  return state;
}

Ynot1, 2018-04-25:
Cant use sinric, its not released in NZ. But its not the skill i am having an issue with, its the values returned to Alexa in the ESP-2866 code.

I want to return the state of GPIO2 (which i have read into the booleen  variable "GarageDoorState") to Alexa so it appears as the smart switch state.

"I am not sure what are you trying to do. Check whether sinric.com can
support"

Ynot1, 2018-04-28:
Anything is possible given time and perseverance!

I had to hack switch.cpp about to do it, but Alexa is now getting her state of my garage door from the door position switch itself, not from the relaystate variable that tracks if the last command was an On or off.

I had to expose the garage door state to switch.cpp via an extern bool declaration ( GarageDoorState is defined in the main.ino) and then fire that Boolean state into the getbinarystate logic ( I just overwrote switchstatus)

This is really crude, but as I only have two controls on that ESP-01 (door open and door close) having the same door state sent back for both of them makes sense. It wouldn't work if you wanted to have several different inputs pins reporting for different functions.

While I was about it, I changed the door state to be read from the rx pin instead of GPIO2 - now the ESP-01 can startup reliably with  the garage door either open or shut. Previously, it would hang if it started with the door open.