kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #49: arduino-esp8266-alexa-multiple-wemo-switch

Ade9000, 2018-04-25:
hi, i am trying to either add a new switch or rename an existing switch, however whatever i try, the code refuses to compile, and jut returns an undefined reference to the item i m trying to add/alter!

i dont have issues with switching the devices already added, i just want to customise it to my requirements. 

please any help would be greatly received, and stop me from actually going insane!!

thank you in advance

Ynot1, 2018-04-25:
Have you tried editing and recompiling something simple, like blink?

Ade9000, 2018-04-26:
> Let's leave Adding a new device till later...
>
> Starting with the code you pull from git, and USING THE FIND REPLACE FUCTION, (not your eye) swap "light" to "plug" or something else. Save, recompile.
>
> Rinse and respeat making small changes each time along the way.

Hi, 
Thought that maybe you had cracked it!
I made the changes using find replace, and this time the code did compile, however when I load the code to the Wemo device and open the serial monitor I am seeing exception (28):
And a load of hex code and eventually connection failed :(

Ade9000, 2018-04-26:
Hi, just found that I hadn’t cleaned up my last attempt properly!! Sorry!

That has worked for re-naming!!
Thank you very much :)

Would still like to add a new device though please if you don’t mind helping :)

Ynot1, 2018-04-27:
Assuming the original code you would search for "kitchen" and every time you find it, copy the whole line you found it on to the line immediately below and change your copied line to whatever you want the new device you want to be called. 
As you have swapped kitchen to something else already , you will probably be searching for whatever you changed kitchen light to.
This will add about 6 lines of code to the top part of the code. The other change you should make is to tweak the local port number for your new device to 83 in your copy of line 45. 

Copy the whole chunk of each of the two 4 line functions around the middle of the code ( lines 77-89 in the original) that mention kitchen several times, and again, change kitchen to whatever every time you find it in your copy.  Also change switch 2 to 3 in your copied functions.

The code as released isn't going to actually close any relays of course, it's up to you to define which processor pins you want to use for that, make them outputs, and then put digital write commands into those on and off functions.

Good luck.