kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #51: Problems running more than 1 instance of this code

Ynot1, 2018-05-27:
I was trying to run 2 instances of this code, on seperate esp-01's so I could test the garage door controller I was building for a friend, whilst keeping my own garage door control going.

The esp-01's were doing dhcp so had unique ip addrees's, I renamed the Alexa invocation words so they were unique , and set the web ports so there was no overlap on the two devices , but I still couldn't make Alexa happy with both of them at the same time.
Eventually had to switch one of them off to get it stable.

Anyone hazard a guess why?

ksnagesh19, 2018-05-28:
Can you post the code, please to cross check.

Ynot1, 2018-07-08:
Answering my own question from a few months ago
To run more than one instance of this code, at least the serial number in switch.cpp needs to be unique, or Alexa loses touch with one of the devices after a while.
I fiddled with the01-NLS line in the xml and made that unique as well. All good now.

ksnagesh19, 2018-07-08:
Thank U

kakopappa, 2018-07-08:
@Ynot1 can you upload the fixed code?

Ynot1, 2018-07-08:
I have hacked that code so much while customising it into a specific application that i hesitate to do that. I will upload the 3 changed lines as pseudo code in a day or to.

Ynot1, 2018-12-16:
I have made more discoveries with having multiple instances of this code running in a single network, there are several places in the code that values have to be unique in the same LAN.

In the main sketch make this unique:

local Port numbers in the command that defines each switch
// Format: Alexa invocation name, local port no, on callback, off callback, Alexa talley indicator 

Alexa wont discover devices whereduplicate port numbers are used

in switch.cpp make all these unique:

sprintf_P(uuid, PSTR("38323636-4558-4dda-9188-cda0e6%02x%02x%02x"), // as released by kakopappa
"<serialNumber>221517K0101769</serialNumber>" //as released by kakopappa
"01-NLS: b9200ebb-736d-4b93-bf03-835149d13983\r\n"  //as released by kakopappa

Although Alexa will initially discover the devices if these arent unique, she will only maintain control of the last device found.

And, to save him the trouble of adding the comment himself, kakopappa would say "Use Sinric skill instead to solve all these issues"