kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #52: Alexa won't find more than 5 switches...

Ynot1, 2018-07-08:
If I define more than 5 switches, Alexa randomly fails to discover some of them. All the switches I have defined work fine idenpendanty, but discovery of all of them is a problem. The code says "max 10" - what limit is this, and have I hit it early?

EWdG, 2018-08-10:
I came across the same problem.
From the sixth switch Alexa will not find a new device. Also, Alexa does not find a previously known switch when asked to "switch one on".
Has anyone defined more than 5 switches successfully? If so, how?

santosh09142, 2018-08-10:
I am also struggling for same, If any one have  solution please help us. I am giving less preference to Sinric as i dont want to control power from internet(security reason)

I have seen after adding 5+ switches script start crashing(reboot nodemcu) and in VS  2017 i seen error stating some code trying to access cpu private info(in nodemcu). If you want i can share error.

EWdG, 2018-08-12:
The basis of this sketch is the code of kakopappa "https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch".
I only expanded it by defining the outputs. In the above sketch, if all lines addressing the sixth output "Vf" are cleared, the 5-output sketch runs very reliably.
It may be that Alexa does not recognize all 5 outputs at the first scan. However, a second search leads to the desired result and all 5 switches have been found.
I suspect the bug that no more than 5 switches work flawlessly, somewhere in kakopappa's programs. I think we have to wait until the programmer has time to take care of the matter. He indicated this on 18 July 2018 at "https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/54".
santosh09142: I understand you so much that you did not get more than 5 switches to run. Why do you suppose that from the sixth switch, the modemcu would always reboot after a while?

Ynot1, 2018-08-12:
I found the same issue - discovery of new Alexa devices becomes more unreliable as more switches are added to the multiple-wemo code. 4 was my sweet spot.
Might be a timing issue related to the small 45 sec or 1 minute period that Alexa spends searching for new devices - there is A LOT of traffic on the debug screen for most of that period with 4 switches enabled.