kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #53: Relevant Alexa bug? Discovery problems...

Ynot1, 2018-07-08:
This may be root the cause of my Alexa not finding >5 switches yesterday .

After a few hours of quite intensive device discovery, my Alexa stopped finding any new devices . I could see XML being uploaded from the multiple switch code when the discovery was underway , but Alexa app didn’t display any result. Put the code onto a new esp-01 with the same result.
In desperation I reset the echo speaker and tried discovery again. Alexa then had 2 copies of each switch ! Looks like discovery was happening all along , but that the results weren’t being saved somewhere.

kakopappa, 2018-07-09:
Why not use the skill I made? This is one of the reason I made it

kakopappa, 2018-07-10:
well, it's available in US, German, India, UK, Canada, Australia and New Zealand