kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #54: Basic lack of understanding.

EWdG, 2018-07-16:
It lies hidden in my mind, as I can address the "WeMos D1 mini" PIN D0 to D8 in the example wemos.ini to perform switching functions.
Since 3 days, after the purchase of an Amazon Echo Dot, I now deal with the topic of signal exchange.
At Sinric.com I opened an account and defined a switch. I got the sketch wemos.ini running on the "WeMos D1 mini" and Alexa recognizes the two switches specified in the example. When I talk to Alexa I can see in the monitor that the two switches are also switched on or off.
Only which PIN of the "WeMos D1 mini" is behind it? Nothing happens at the exits. In the sketch I also find no definition of an OUTPUT. Have I overlooked something fundamental?

EWdG, 2018-07-18:
Hello Aruna.
Since I did not receive an answer due to my request from two days ago, I ventured myself to the cause.
Based on your sketch sinric.ino (for a switch also with several WeMos he runs very well) I made entries in your wemos.ini sketch.
...
const int relayPin1 = 5; // D1 pin
const int relayPin2 = 4; // D2 pin
...
...
  pinMode (relayPin1, OUTPUT);
  pinMode (relayPin2, OUTPUT);
...
Your free line below:
bool officeLightsOff () {
  Serial.println ("Switch 1 turn off ...");
  (... free line ...)
  isOfficeLightsOn = false;
  return isOfficeLightsOn;
I have supplemented by:
digitalWrite (relayPin1, LOW); // turn off relay with voltage LOW
Accordingly, I have done this in the other 3 sections.

And behold -> it worked the same on my nodeMCU 1.0.
Thanks again for this good sketch.

If you could find time to add new hints besides your new projects sinric.com and the work at https://github.com/kakopappa/sinric here, it would be easier for newcomers like me.
In your new project, I will also work in pure.

closed 2018-07-18 by EWdG

kakopappa, 2018-07-18:
sorry i missed your question somehow. I will add a new sketch for WeMos D1 with Relay