kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #55: Relay Reverts To Previous State

eddcaton, 2018-08-01:
Hey, 

I have this code working, but have found an issue.

When i use alexa or the alexa app to turn the light on or off the relay will change but then will change back to its original state after one or two seconds.

It seems like there is a fair amount of chat about this issue throughout the internet, but no one has posted a fix to the issue.

Would anyone on here have a solution to this issue?

Ynot1, 2018-08-02:
> @eddcaton<https://github.com/eddcaton> may be it's sending SetBinaryState two times?
>
> think about moving to sinric skill.
> https://sinric.com
>
> this is why i made this skill. amazon keep on changing the way firmware work

I am using a branch of that Code for several alexa based projects and they are all ticking along nicely - nothing has changed in Alexa land that has broken it.

Hey edcatton - there is lots of nice logging going on if you look at the serial port display  - does that give you any hints about if on or off controls are being interpreted ?

LZHenry, 2018-08-02:
This code shows that ALEXA does interesting things, like repetitive commands.

1. I have this code implemented with 12 devices
2.ESP8266 gets ALEXA issued command
3. I use ARDUINO Nano 328 to convert it into "X10" commands to interface with the rest of my existing system.

 
Interesting....
When I open the ALEXA APP on my iPhone, and just LOOK at the devices... the APP CONTINUALLY issues communications;  these we can see using the print statements to my ARDUINO monitor.  Just discovered this week and have not invested time looking at the commands.  Similar strange commands occur when opening the ALEXA application on the PC and simply LOOKING at the Smart Home device screen... ALEXA commands a device.

This may not be the same as you are seeing, but this is ALEXA doing strange commands on our home network... we never would know if it were not observed with this code implementation.

@kakopappa do you already know what these transmissions are?????

LZH