kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #57: Alexa not sending to device?

jenniferannegluck, 2018-09-20:
I had everything working -- for a couple hours --

Alexa found the switches/devices.  (I had three "Door", "Open Close", and "Elevator Up")  I told Alexa to turn on/off for the different devices, and she did it.  I had put a couple print statements in the code so I could see it working, too.  I just left it all on and running.

In a couple hours it wasn't working.  Alexa now says "Door isn't responding, Please check it's network connection and power supply"  So, I hooked it back up to the computer to watch the Serial output.

It scrolls through a lot of this (took out the IP address and port number)...

Sending response to xx.xx.xx.xx
Port : xxxxx
Response sent !
Got UDP Belkin Request..

And also this...

 ########## Responding to setup.xml ... ########

Sending :<?xml version="1.0"?><root><device><deviceType>urn:Belkin:device:controllee:1</deviceType><friendlyName>open close </friendlyName><manufacturer>Belkin International Inc.</manufacturer><modelName>Socket</modelName><modelNumber>3.1415</modelNumber><modelDescription>Belkin Plugin Socket 1.0</modelDescription>
<UDN>uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e6b40a51-85</UDN><serialNumber>221517K0101769</serialNumber><binaryState>0</binaryState><serviceList><service><serviceType>urn:Belkin:service:basicevent:1</serviceType><serviceId>urn:Belkin:serviceId:basicevent1</serviceId><controlURL>/upnp/control/basicevent1</controlURL><eventSubURL>/upnp/event/basicevent1</eventSubURL><SCPDURL>/eventservice.xml</SCPDURL></service></serviceList></device></root>

So, I know that it's still connected to the WIFI.  But when I give Alexa the command, nothing happens.  The WEMO doesn't seem to respond.  (It just continues to scroll through the same stuff as above.)  I am a newbie, so speak in small words if you happen to know the issue and can help.

Ynot1, 2018-09-20:
Tried a Reboot of the Alexa device?

Sending :urn:Belkin:device:controllee:1open close Belkin International Inc.Socket3.1415Belkin Plugin Socket 1.0
uuid:Socket-1_0-38323636-4558-4dda-9188-cda0e6b40a51-85221517K01017690urn:Belkin:service:basicevent:1urn:Belkin:serviceId:basicevent1/upnp/control/basicevent1/upnp/event/basicevent1/eventservice.xml

jenniferannegluck, 2018-09-20:
Ha!  Thank you -- it worked.

I also should have read the earlier threads.  I have two WEMO's running.  So, I changed the serial number and the 01-NLS as suggested by (I guess you in an earlier thread).   That didn't work immediately -- but when I rebooted Alexa it started working again.  Now we'll see if it continues working.  Hopefully I won't have to keep rebooting Alexa, otherwise it will make the application pretty useless.

THanks for the help... do you think it was the reboot or the serial renaming that did it?

Ynot1, 2018-09-20:
If you had two wemos with same serial numbers  ect, then whichever is seen by Alexa last will work , the other would eventually fail.
Just changing one of them doesn’t solve the problem till she who will not be named is restarted as well.

Aye - that was me, glad it helped someone else.

I have 3 wemos all running happily now using same process.

jenniferannegluck, 2018-09-21:
Tried again today -- and no go... same issue.  Rebooting Alexa didn't work.  Fiddling with serial numbers again didn't work.

So, here's my theory.  I have 3 alexa's.  I think that they remembered the old serial number on the device.  I went to the Alexa app on my phone and deleted the devices (not sure how else to delete devices).  But, that must not have completely deleted it (when I asked Alexa to turn on the device she still tried even though I had deleted it.)  So, I unplugged the other alexa's in the house until when I asked her to turn on the device she said she didn't recognize it.  

Reloaded the code with the new serial number etc. onto Alexa.  Plugged them in.  Now it is seemingly working again.  

We'll see if it continues to work or if I get the same issue tomorrow.

jenniferannegluck, 2018-09-22:
It has again failed.

Here is my setup:
1.  Three separate echos in different rooms
2.  WEMOs with one switch (that works just fine).
3.  WEMOs with three switches (that works, then starts not receiving Alexa's callback after a day).

So, the switches all have a different port.  (I assigned port 80, 81, 82 to the WEMOs with 3 switches and 83 to the WEMOs with one port.)
The two WEMOs have a different serial number and 01-NLS number.  I kept the format of these, but just changed some random numbers in them.

I have no idea what to try next.

kakopappa, 2018-09-22:
Sinric?

Ynot1, 2018-09-22:
Bugger.

I am now racking my brain to recall if there was anything else I changed in the code to make 2 instances of it stable. I said I would upload pseudo code snips of what I did change in switch.cpp, but I can't see that I actually did that it in the end. Maybe I put it in another issue thread?
The changes i made isn't in github, or available  to me for the next few days.
But my setup, (which is the same as yours apart from only having a single Alexa ) has worked for several months, with regular use of all switches

jenniferannegluck, 2018-09-23:
I am trying one more thing...  I am also changing the uuid.  So far, no luck, but I think it might be because the old device hasn't gone away, which leads me to ask...

 I am trying to get Alexa to forget this device so that I can re-find it with the correct uuid/serial #/etc...  But how do I get her to do that?  I tried rebooting, but she still recognizes the devices.  

Somehow before I managed to delete and then re-find the devices, but it's not working this time.

Ynot1, 2018-09-24:
Ohh - the uuid -,maybe I changed that as well? I think that is auto generated from the MAC Address though.

Log into the Alexa web interface - delete devices there. I got inconsistent results doing that via the Alexa app.

santosh09142, 2018-09-24:
Else you can reset exlea ....

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/57#issuecomment-423858062",
"url": "https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/57#issuecomment-423858062",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [kakopappa/arduino-esp8266-alexa-multiple-wemo-switch] Alexa not sending to device? (#57)",
"sections": [
{
"text": "",
"activityTitle": "**Ynot1**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@Ynot1",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"kakopappa/arduino-esp8266-alexa-multiple-wemo-switch\",\n\"issueId\": 57,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"kakopappa/arduino-esp8266-alexa-multiple-wemo-switch\",\n\"issueId\": 57\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/57#issuecomment-423858062"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 382321410\n}"
}
],
"themeColor": "26292E"
}
]

jenniferannegluck, 2018-09-26:
Ok, I had no idea what the comment Sinric was.  Now I understand it is some kind of alexa skill to use the WEMOs with.  Can you @kakopappa give a brief explanation of how to switch to using that?  

Would I use this code in conjunction with that?  Or would I completely replace this code with using sinric?

kakopappa, 2018-09-26:
@jenniferannegluck 

everything you need to know in the home page
https://sinric.com/