kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #61: Echo Dot Gen 3 Compability

fhschmitt85, 2018-11-01:
The code itself seems to work fine with Amazon Dot Generation 3. There are no problems adding new devices in the Alexa App. 
But there are apparently communication probelms. If you are giving commands for switching operation, the MCU runs the callback functions and sends  the proper xml file to Alexa. Alexa did not response immediately and states finally that there is a problem with the device. After some getBinaryState requests the Alexa App shows the right device state.

Does anyone have an idea dealing with this issue?

kakopappa, 2018-11-02:
I have seen this js happening in latest firmware updates from Amazon.

Have you tried using the Sinric Skill? Sinric.com

fhschmitt85, 2018-11-02:
Not yet.
I'm going to write an Alexa skill using AWS IoT API and an esp8266 MQTT implementation.
Therefore I can integrate different, more complex devices

pmancele, 2018-11-28:
@fhschmitt85 Using skills and AWS IoT is also my plan, as MQTT will be usefull for all kind of devices !
But i understood that when creating a skill you have to use its name before each Alexa command ..
Like if you design a connected switch using AWS IoT and your skill you have to tell "Alexa, ask {my_skill} to turn on the switch"
Is it true or do you have an idea to make your items recognized as smart home object ? So it will be possible to say directly "Alexa, turn on the switch"

kakopappa, 2018-11-28:
If you are using a smart home skill, you don’t have to use the skill name
in front. You can just say Alexa, turn on the tv

fabio14, 2019-02-15:
I had the same issue with 3 gen dot echo, commenting several Serial.println (above all in the UPNPBroadcasterResponder file) improved Alexa's response speed a lot

ArkajyotiChatterjee, 2019-03-26:
Hello,
I have an Echo Dot Generation 3. The same code was working flawlessly with Echo Dot Generation2 but the 3rd generation seems not to find / discover any devices.
The Smart Home Skill I am using is YeeLight ( both in Gen2 and Gen3).
Any help would be appreciated.
Thanks,
Arka

kakopappa, 2019-03-26:
Have you tried Sinric skill?

ArkajyotiChatterjee, 2019-03-26:
Hello Aruna,
I wanted to use this multiple wemo code. So any way to make that happen?
Thanks ,
Arka

kakopappa, 2019-03-26:
This is kind of the reason why I made the skill. New Alexa updates stopped
sending the UDP unicast on the network

Do you see the Alexa UDP requests in the Serial terminal? If not using the
skill is your only option

sourabhjaiswal, 2020-07-15:
Hi,

I am also facing the same issue. I have enabled sinric skill also.
Anyone got any solution to make it work.

Thanks,
Jaiswal

sourabhjaiswal, 2020-07-16:
Hi Kakopappa,

Thanks a lot for the update.
Actually I am just starting with IOT and not looking for any paid platforms as of now.
I will surely try sinric.pro with initial no cost 3 devices.
It will be a big help if you can suggest some other platform.
Suggestion for guides or tutorials are highly appreciated.

Regards,
Jaiswal.

kakopappa, 2020-07-17:
try Home Assistant or domoticz