kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #68: Connection problem between ESP3266 and alexa on amazon Echo

yannicktremblay3, 2019-08-13:
Hello,

I bought a Amazon echo last week and I did some test with my ESP3266 to eventually make a project... Last week all is worked but this week my amazon echo can't find my ESP3266 on my network, I did no change since last week... I tried to reset my network, reset my echo, put some program that worked lasw week on my esp but it always cant find my esp... I tried an other esp3266 module and had the same problem. Someone can help me?

 

Thanks

kakopappa, 2019-08-14:
Use https://Sinric.com

Ynot1, 2019-08-14:
You are only trying to connect a single esp device right? - that code doesn’t scale up without some minor tweaks to make each device unique.

Otherwise, Delete the device from your amazon account and try discovery again.

yannicktremblay3, 2019-08-15:
kakopappa, I tried sinric but it can't detec my device anyway.
Ynot1, I deleted my ESP to my account I can't see it on my connected device but when I try to detect it's always the same problem.

yannicktremblay3, 2019-08-16:
I finally find the problem, I changed my wi-fi rooter  and it solve the problem. But I dont understand why my last rooter blocked the connection... it have 3 internal antenna but I also tried to connect all of my device on the same 2.4ghz antenna (because we can't connect ESP to 5 ghz) and the problem was the same. I thinked that if all of my device is on the same network and same antenna it could resolve the problem but no... someone have some explain?

yannicktremblay3, 2019-08-16:
I tried an other thing with my older rooter, i put all of my ip adress device (amazon echo and esp) in reserved (fix)  and changed them to other ip and it work i can find my esp.

closed 2019-08-16 by yannicktremblay3

Ynot1, 2019-08-16:
Some old routers have a very small dhcp pool, and so can’t give out an IP address. I had a similar problem when I added many esp devices.

Closed #68<https://github.com/kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/68>.