kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #69: Not an Issue But a question

rajdpandey, 2019-09-11:
Hi,

In your code at this stage:
 office = new Switch("office lights", 80, officeLightsOn, officeLightsOff);
 kitchen = new Switch("kitchen lights", 81, kitchenLightsOn, kitchenLightsOff);

you are referring invocation as "office lights" & "kitchen Lights" and then port and following that 
"On callback" and "off call back".

So here in On call back you are turning something on and in off turing it off.
So let's say example I have created the same for TV :
TV= new Switch("TV", 80, TVOn, TVOff);  // This is for tv turning on and off

So What should I write next to increase, decrease or mute the TV volume?
Should it be like this:
 Volume= new Switch("Volume", 81, Increasevol, decreasevol);
 mute = new Switch("Volume", 81, mutevol, unmutevol);

and respective function for Increasevol, decreasevol, mutevol & unmutevol I will create.
Please help regarding above...

kakopappa, 2019-09-11:
emulator supports only on/off. 

If you want more functionality, checkout the sinric.com project

rajdpandey, 2019-09-11:
Yeah, I checked that but I don't want to use any third-party things I just want to Implement this by direct communication between Alexa and Nodemcu or If there is another method other than using sinric.com.

crwhite57, 2021-03-25:
I don't want to use Sinric because they charge you for each device over 3 devices, besides I got this to work "kind of" with my gen2 dot. The only problem is that it says that we can have 10 devices but my echo will only find 5. I am using an ESP8266 D1 mini and I am programming it as a Wemos D1 R1. I have ordered an actual Wemos D1 R2 and I will try this with that but until someone comes up with a way to fix this restriction problem, I have found a workaround.

If you need more than 4 devices, program multiple chips. The first one would be programmed with 4 devices using ports 80-83, the second the same but use ports 84-87, and the third with ports 88 and 89.

When I get the new Wemos D1 I will test this sketch on it and let you know if it works better on that chip.

Ynot1, 2021-03-25:
I found exactly the same issue - 4 devices is all I could get to work in a stable fashion. I think the constraint is more in Alexa land than the local processor , but that’s just a hunch

crwhite57, 2021-06-27:
I found the solution to this problem. I found that if you have this in your sketch:

void loop()
{
  espalexa.loop();
  delay(1);
}

The delay is too quick and Alexa does not discover everything, or not correctly. The delay(1) cycles it too fast for Alexa. Change the 1 to 10. Making that one change, now I see and operate 8 devices on my WeMos D1 Mini and D1 R2. With my ESP32 I have 16 devices. If you want to know more, check my youtube video at
https://www.youtube.com/channel/UCOGIb90EIX9c0AdaNbF_yjA