kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #71: I can't use the skill for this project. When I ask Alexa to discover I get the following

simmtek, 2020-02-02:
Exception (28):
epc1=0x4020b2d2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3ffffce0 end: 3fffffd0 offset: 01a0

>>>stack>>>
3ffffe80:  00000060 00000060 00000000 40100575  
3ffffe90:  40104ab2 088cadf2 3fff23b0 40208062  
3ffffea0:  40104b6e 3ffefaf8 088cadf2 3fff03e4  
3ffffeb0:  00000060 3fffff34 3fffff34 4020b80b  
3ffffec0:  3fff098c 0000005e 3fffff34 4020b857  
3ffffed0:  00000000 00000000 3fffff34 4020b889  
3ffffee0:  3ffe8fd4 3fff098c 3fffff34 4020b8d6  
3ffffef0:  3fff098c 00000000 3fff0d54 4020b349  
3fffff00:  3fffdad0 3fff098c 3fff0410 40206420  
3fffff10:  08194cf5 3fff0e2c 00000000 3fff0e2c  
3fffff20:  3fff00f8 3fff03c4 3ffe852d 4020ad30  
3fffff30:  3fffdad0 3fff1354 0000005f 0000005e  
3fffff40:  4020cc88 0901a8c0 3fff0374 4020398a  
3fffff50:  00000400 00000002 00000000 4020c588  
3fffff60:  40204de8 feefeffe 4020c59c 3fff0e2c  
3fffff70:  3fffdad0 00000000 3fff0e24 40205965  
3fffff80:  1a01a8c0 feefeffe feefeffe feefeffe  
3fffff90:  00001fd4 00000000 feefeffe feefeffe  
3fffffa0:  00000000 00000000 00000001 3fff0e2c  
3fffffb0:  3fffdad0 00000000 3fff0e24 4020c344  
3fffffc0:  feefeffe feefeffe 3ffe8668 40100739  
<<<stack<<<

PetRatty, 2020-02-10:
Exception (28)
Don't know how to solve this without more info
Have you changed the stock code? except for the wifi name and password
does your esp boot normally and  discover the devices?
is it only when you ask her to change state of the switches it crashes?
Exception 28 is an illegal stack call or invalid pointer address, you could be changing something while stuck in a loop

some people have said they get exception 28 on the esp8266 and with the same code runs fine on an esp32, others connecting to home wifi it crashes but on the office wifi it's ok, losing internet connection and sending a request seems to be the most common cause, there is a tool called GDB  https://github.com/esp8266/Arduino/tree/master/libraries/GDBStub never used it but it can track and trace all the calls in your program and show you where the error comes from.

simmtek, 2020-02-10:
I code change except WiFi etc, I will try a ESP32 when it comes, as I thought I’d try that anyway

simmtek, 2020-02-10:
Thank you for the reply
It does boot normally and connects to WiFi too, that exception only happens the instant I ask Alexa to discover new device

Regards,
Kevin

Juniiorf, 2020-05-03:
I have this issue

simmtek, 2021-03-07:
I fixed this by changing char packetBuffer[512]; to  char packetBuffer[1024];    
at around line 5 in WemoManager.cpp