kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #74: Add a push button to manually turn on/off the lights

ChompsR, 2020-04-22:
Good evening

I've been working on this project for the last week, so it came this idea to add a push button to manually turn on and off the lights. After studying where this instructions had to be placed, I finally found that is on Switch.cpp (continue reading). 

There's a void called Switch::handleUpnpControl() which I presume this is the one that controls when you ask Alexa to turn on/off the device, this is where I add some code lines to include a push button. 

**RESULTS:**
- I can get Alexa to turn on/off the device.
- The code detects the push button BUT **it only makes changes when the button is pressed and I had to ask Alexa to turn on/off the lights**. I thought this can be because there's some statement that I can not found yet that takes the instruction direct to this void.

Any ideas of how we could add this button to had a fully functional smart device? 

I also added some new lines to the original Wemos page to make the device smarter, I will upload the pull request once the push button is coded.

Things I added:
 - Automatic reset when connection lost or not found.
- Detection for and OLED display to show a little logo, SSID and IP address.

For this changes stay tuned.

This is what I modified on Switch.cpp:

<pre>
<font color="#00979c">void</font> <font color="#000000">Switch</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">handleEventservice</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
 &nbsp;<b><font color="#d35400">Serial</font></b><font color="#434f54">.</font><font color="#d35400">println</font><font color="#000000">(</font><font color="#005c5f">&#34; ########## Responding to eventservice.xml ... ########\n&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
 &nbsp;<font color="#00979c">String</font> <font color="#000000">eventservice_xml</font> <font color="#434f54">=</font> <font color="#005c5f">&#34;&lt;scpd xmlns=\&#34;urn:Belkin:service-1-0\&#34;&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;actionList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;action&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;SetBinaryState&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;argumentList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;argument&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;retval&#47;&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;BinaryState&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;relatedStateVariable&gt;BinaryState&lt;&#47;relatedStateVariable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;direction&gt;in&lt;&#47;direction&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;argument&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;argumentList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;action&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;action&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;GetBinaryState&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;argumentList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;argument&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;retval&#47;&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;BinaryState&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;relatedStateVariable&gt;BinaryState&lt;&#47;relatedStateVariable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;direction&gt;out&lt;&#47;direction&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;argument&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;argumentList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;action&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;actionList&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;serviceStateTable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;stateVariable sendEvents=\&#34;yes\&#34;&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;BinaryState&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;dataType&gt;Boolean&lt;&#47;dataType&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;defaultValue&gt;0&lt;&#47;defaultValue&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;stateVariable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;stateVariable sendEvents=\&#34;yes\&#34;&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;name&gt;level&lt;&#47;name&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;dataType&gt;string&lt;&#47;dataType&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;defaultValue&gt;0&lt;&#47;defaultValue&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;stateVariable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;serviceStateTable&gt;&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;&lt;&#47;scpd&gt;\r\n&#34;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;<font color="#005c5f">&#34;\r\n&#34;</font><font color="#000000">;</font>
 &nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;&nbsp;
 &nbsp;&nbsp;&nbsp;<font color="#000000">server</font><font color="#434f54">-</font><font color="#434f54">&gt;</font><font color="#d35400">send</font><font color="#000000">(</font><font color="#000000">200</font><font color="#434f54">,</font> <font color="#005c5f">&#34;text&#47;plain&#34;</font><font color="#434f54">,</font> <font color="#000000">eventservice_xml</font><font color="#434f54">.</font><font color="#000000">c_str</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">)</font><font color="#000000">;</font>
<font color="#000000">}</font>
 
<font color="#00979c">void</font> <font color="#000000">Switch</font><font color="#434f54">:</font><font color="#434f54">:</font><font color="#000000">handleUpnpControl</font><font color="#000000">(</font><font color="#000000">)</font><font color="#000000">{</font>
[... 43 lines cut ...]
[... cut ...]
 &nbsp;<font color="#000000">server</font><font color="#434f54">-</font><font color="#434f54">&gt;</font><font color="#d35400">send</font><font color="#000000">(</font><font color="#000000">200</font><font color="#434f54">,</font> <font color="#005c5f">&#34;text&#47;plain&#34;</font><font color="#434f54">,</font> <font color="#005c5f">&#34;&#34;</font><font color="#000000">)</font><font color="#000000">;</font>
 &nbsp;
<font color="#000000">}</font>

</pre>

Thank you very much and keep safe.

- ChompsR

kakopappa, 2020-04-23:
> void Switch::handleEventservice(){
>   *Serial*.println(" ########## Responding to eventservice.xml ... ########\n");
>   String eventservice_xml = "<scpd xmlns=\"urn:Belkin:service-1-0\">"
>             "<name>SetBinaryState</name>"
>                 "<name>BinaryState</name>"
>                 "<relatedStateVariable>BinaryState</relatedStateVariable>"
>                 "<direction>in</direction>"
>             "<name>GetBinaryState</name>"
>                 "<direction>out</direction>"
>             "<name>BinaryState</name>"
>             "<dataType>Boolean</dataType>"
>             "<defaultValue>0</defaultValue>"
>               "<name>level</name>"
>               "<dataType>string</dataType>"
>               "<defaultValue>0</defaultValue>"
>         "</scpd>\r\n"
>     server->send(200, "text/plain", eventservice_xml.c_str());
> void Switch::handleUpnpControl(){
>   *Serial*.println("########## Responding to  /upnp/control/basicevent1 ... ##########");
>   //for (int x=0; x <= HTTP.args(); x++) {
> pinMode(Boton,INPUT_PULLUP); // PUSH BUTTON
> Estado = digitalRead(Boton); // STATE
>   String request = server->arg(0);
>   *Serial*.print("request:");
>   *Serial*.println(request);
>   if(request.indexOf("SetBinaryState") >= 0) {
>     if(request.indexOf("<BinaryState>1</BinaryState>") >= 0) {
>         *Serial*.println("Got Turn on request");
>         switchStatus = onCallback();
>         Estado = digitalRead(Boton);
>       if(Estado != HIGH){
>         request.indexOf("<BinaryState>0</BinaryState>") >= 0;
>         *Serial*.println("Got Turn off request");
>         switchStatus = offCallback();
>     if(request.indexOf("<BinaryState>0</BinaryState>") >= 0) {
>         request.indexOf("<BinaryState>1</BinaryState>") >= 0;
>   if(request.indexOf("GetBinaryState") >= 0) {
>     *Serial*.println("Got binary state request");
>   server->send(200, "text/plain", "");

Thanks for sharing. FYI Emulator technique time to time cause problems. I
have switched to amazon skill based solution now using https://sinric.pro

ChompsR, 2020-04-23:
Kakopappa, thank you very much! Sinric Pro skill is amazing, and the codes/libraries/examples, just wow, I'm really impressed, if you don't mind I will make some changes to the Switch example just to add the lines that I've already coded on wemo's example. 

I'll be in touch!

closed 2020-04-27 by ChompsR