kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #78: can i change invocation phrase?

joortiz22, 2022-06-15:
I'd like to use this to open and close my windows and doors. how can i change "turn on" to "open" and "tur off" to "close"

kakopappa, 2022-06-15:
Not possible. Emulation supports only on/off

You can use sinric pro skill or set up custom phrases to do this

Ynot1, 2022-06-15:
Create an Alexa routine and choose “open window” as the custom speech to trigger it. Then as the action choose “turn on” the device.
Although the routine triggered whenever I spoke the words, it only operated the output sometimes. I got 100% reliability by choosing the custom output option as well and getting it to simulate me saying “Alexa turn on the lounge curtain”

I just motorised my lounge curtains using this exact same code and using this process .