kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #79: Question on Alexa status

Jamesb4uk, 2023-11-04:
Bit of a long shot, how does this control the way Alexa sees the relay is on?

I'm using this fine with a relay which pulses a 12v diesel heater switch.

The relay is connected on the back of the on/off switch. To power on it activates the relay for 1 second then turns off the relay  & the heater starts up, then turning off it activates the relay for 5 seconds to begin the shutdown sequence of the heater & turns the relay off. the Alexa switch status only stays on when the relays on.

Is it possible to keep the status as on (even when the relays off) so i can see when the heaters running?

Kind Regards
James

Ynot1, 2023-11-04:
It doesn’t. Alexa only sees the relay activation status, not the state of whatever you attach to it

But I hacked around one of those two other functions that are compiled along with the main Audino sketch to achieve this by assigning another auduino pin to be the feedback towards Alexa  instead. Took me ages and only works for a single control , but it’s functional , sort of.

You still can’t ask Alexa if the thing is on or off, but at least it displays correctly in the app if you look at it

But if you

Jamesb4uk, 2023-11-04:
Thanks Ynot1

I started reading some of the other comments on how this interacts with Alexa & just built a 2nd test device to play with.

Hooking up the serial monitor is proving useful as i can see the comms to the D1 Mini

Got UDP Belkin Request..

Sending response to 
Port : 60562
Response sent !
Got UDP Belkin Request..

Sending response to 
Port : 42294
Response sent !
########## Responding to  /upnp/control/basicevent1 ... ##########
request:<?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/"s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:SetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:SetBinaryState></s:Body></s:Envelope>
Got Turn on request
Relay On
Relay On
Relay On
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>1</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

########## Responding to  /upnp/control/basicevent1 ... ##########
request:<?xml version="1.0" encoding="utf-8"?><s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/"s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body><u:GetBinaryState xmlns:u="urn:Belkin:service:basicevent:1"><BinaryState>1</BinaryState></u:GetBinaryState></s:Body></s:Envelope>
Got binary state request
Sending :<s:Envelope xmlns:s="http://schemas.xmlsoap.org/soap/envelope/" s:encodingStyle="http://schemas.xmlsoap.org/soap/encoding/"><s:Body>
<u:GetBinaryStateResponse xmlns:u="urn:Belkin:service:basicevent:1">
<BinaryState>1</BinaryState>
</u:GetBinaryStateResponse>
</s:Body> </s:Envelope>

Got UDP Belkin Request..

Sending response to 
Port : 45728
Response sent !
########## Responding to  /upnp/control/basicevent1 ... ##########

I think the Binarystate is what I'm going to work on figuring out as when it turns of it switches to 0

I'll post back if i work it out

Kind Regards
James

Ynot1, 2023-11-05:
> request:<s:Envelope xmlns:s="
> <kakopappa/arduino-esp8266-alexa-multiple-wemo-switch/issues/79/1793565886
> @github.com>

in module switch.cpp

I defined an external boolean

extern bool GarageDoorState; (this is attached to a read of another arduino
pin in the main sketch)

===-===--

I then just crudely overwrote switchStatus with that value before it is
sent to Alexa (which is why this solution isnt great if you are dealing
with more than 1 external thing)

if (request.indexOf("GetBinaryState") >= 0)  {

switchStatus = !GarageDoorState ;
....

Jamesb4uk, 2023-11-06:
Thanks again.

I was actually looking at switch.cpp & going down the road that you have done last night its a on/off switch so no issues for me.

Interesting, you did a not operation on GarageDoorState, maybe its Monday brain shock how does that work with switchstatus?

James

Ynot1, 2023-11-06:
GarageDoorState; (this is attached to a read of another arduino pin in the main sketch)

Jamesb4uk, 2023-11-06:
ok i think i understand, will give it a try.

Appreciate your help in this

James