kakopappa/arduino-esp8266-alexa-multiple-wemo-switch #81: Unresponsive devices…

Ynot1, 2025-04-16:
Has anyone noticed these emulated wemo devices becoming less reliable in Alexa land recently? I have created 10 or so instances of this code over the last 4 or 5 years ( garage door, burgular alarm interface, cat feeder, and more) and I now have a problem where some of them are reporting as unresponsive. I am sure the OP would like me to move to sinric, but I love the approach of emulating an already supported device (and avoiding on going costs). Is it just me, or have other users seen this? This week, both my cat feeder and garage door have been absent….(luckily the cat feeder has a backup timer in the code or pussy would be complaining loudly!)
I noticed the same thing maybe a year ago, but repeated reboots and rediscoveries sorted it out. A bit more of a sticky problem this week. 
Is Alexa trying to drop support of these devices maybe?

kakopappa, 2025-04-16:
Could be related:
https://www.theregister.com/2025/03/17/amazon_kills_on_device_alexa/

tvixen, 2025-06-02:
Yep... thats why I have dropped my Alexa.

Ynot1, 2025-06-02:
That seems a harsh response. Lots of ways around the issue being discussed